balena-io/etcher #4083: Skip validation option before validation screen

probal31, 2023-07-04:
Hello Team,

I use Ether for flashing rpi images to my sd card quite often. 
I normally skip the optional validation step when it reaches the particular step.

However, I feel it would have been great if a checkbox was made available to skip validation in the initial config screen or in the flashing screen or in both. Something like this as in the image below.
[image]
[image]

Can this feature be considered in the app?
I can also work on that but I may need some initial understanding on the `components`, `models`.

I will also give that a try myself but I wanted to raise a feature request first and understand from you how you want it to be.

Thanks.

lurch, 2023-07-05:
There used to be an option for it in the settings, but the Changelog says that it was removed a couple of years ago https://github.com/balena-io/etcher/blob/master/CHANGELOG.md#v15111

fuweichin, 2023-12-10:
The "Skip" link button doesn't work, it just cannot stop validating.

Etcher for Windows 1.9.11