balena-io/etcher #4088: User did not grant permission

ShalevG114, 2023-07-08:
- **Etcher version: 1.18.8** 
- **Operating system and architecture: windows** 
- **Image flashed: Kali Linux latest iso** 
- **What do you think should have happened: flash the usb drive** 
- **What happened: I tried to flash the kali linux iso to my usb to install on my laptop but when i ry to flash it i get this error: 
[image]
**
- **Do you see any meaningful error information in the DevTools? -----> 
[image]
**

ShalevG114, 2023-07-08:
**I DID OPEN IT IN ADMINISTRATOR**

dfunckt, 2023-07-12:
If you don't open it as an administrator, does Etcher ask you for permission when you click on "Flash"?