balena-io/etcher #4094: Flash Failed

Vivek1078, 2023-07-31:
- **Etcher version:** 1.18.11
- **Operating system and architecture:** Win11 64bit
- **Image flashed:** Parrot Security 5.3
- **What do you think should have happened:** I was unable to flash my 8gb usb with the iso file as it was throwing me flash failed error.
- **What happened:** 
[image]

- **Do you see any meaningful error information in the DevTools?**

WallyDW, 2023-10-16:
Maybe same issue here.
I flash my SDCard and the writing seems fine, but then it tries to validate the writing and during that process Windows 11 seems to be discovering the newly written image and refresh the storage list by which Balena Etcher fails on the verifying part.
The image seems to work though.

ianflegg, 2024-01-16:
I get the same error every time after I try to flash an ISO to a USB stick. I have now tried 5 different USB sticks, having purchased a new one today from Amazon. Why does this not work please?