balena-io/etcher #4096: Debian Testing has removed gconf2 packages.

glerroo, 2023-08-07:
- **Etcher version:** 1.18.12
- **Operating system and architecture:** Debian Gnu/Linux Testing
- **Image flashed:** N/A
- **What happened:** Install filed because dependency is not satisfiable: gconf-service

gconf2 is used for gtk2 application and is an archived project.
https://gitlab.gnome.org/Archive/gconf 

modern application like gtk3 and gtk4 use dconf.
https://gitlab.gnome.org/GNOME/dconf

wc1376c, 2023-08-17:
https://github.com/balena-io/etcher/issues/2019#issuecomment-363366530

And here we are in 2023

lurch, 2023-10-04:
Looks like this is related to https://github.com/electron/electron/issues/2727 so maybe this problem will be "automatically" fixed when Etcher moves to a newer version of Electron?

freerko, 2023-11-21:
just ran into this as well :/

XrayBravoGolf, 2023-12-08:
> Looks like this is related to [electron/electron#2727](https://github.com/electron/electron/issues/2727) so maybe this problem will be "automatically" fixed when Etcher moves to a newer version of Electron?

What version if Electron is being used now? Electron [removed gconf 4 years ago](https://github.com/electron/electron/pull/19498); the change happened in Electron [v7.0.0-beta.1](https://github.com/electron/electron/releases/tag/v7.0.0-beta.1). 

As of today, the latest non-beta of Electron is v25. Unless I'm making incorrect assumptions, hard to imagine Etcher is using Electron older than v7.

lurch, 2023-12-08:
@XrayBravoGolf Ahh, good sleuthing! :detective: 
In that case, I don't know why Etcher still needs gconf2 - I was only providing my best guess as an "outside observer" :slightly_smiling_face:

Gibigbig, 2023-12-17:
Just using the Appimage worked for me. Shame the deb is broken

theofficialgman, 2024-01-28:
Its a shame that etcher continues to be unmaintained. it took 1 year and 4 months for this bug https://github.com/balena-io/etcher/issues/3647#issuecomment-1249473908 to be resolved so that etcher could launch on modern disros

looks like after approaching 6 months this current issue is still unfixed that prevents etcher from even being installed on modern disros. guess we will have another year to wait until maintainers fix this one too.

rautamiekka, 2024-02-06:
This is still happening with 1.19.4 ...

qwertychouskie, 2024-02-12:
Also affected by this issue.

ayubowan2u, 2024-02-15:
> Also affected by this issue.

+1

Had to go back to 1.7.9 which still works

teward, 2024-03-14:
The Balena Etcher package definitions state to include `gconf` - see https://github.com/balena-io/etcher/blob/37618ce2fd7cdbc8c26d0fdb4e91f393acfacbab/package.json#L109 and the line after it.

If these were removed or replaced with dconf-specific pacakges instead, this would fix the .deb files.  This requires, however, that the Balena Etcher team determines if there are any hard dependencies on the now deprecated `gconf` components.

This is independent from Electron upstream changes.

theofficialgman, 2024-03-17:
> the now deprecated `gconf` components.

correction. it is not "now" deprecated. It was deprecated in 2014 (10 years ago). https://wiki.gnome.org/Initiatives/GnomeGoals/GSettingsMigration
It is now finally being removed after 10 years of deprecation by OSs since this has given ample time for developers to remove this dependency from their applications. Etcher has not acted on the 10 years of deprecation notice.

jiapei100, 2024-04-28:
Same thing for **balena-etcher_1.19.16_amd64.deb**. Still **NOT** fixed yet...

McxCZE, 2024-05-10:
Jesus holy mary, still not fixed...

fantauzzi, 2024-08-29:
Just reproduced the issue on Ubuntu 24.04.1

theofficialgman, 2024-10-17:
@aethernet this bug can be closed https://github.com/balena-io/etcher/commit/2ed779ef371db367e4e413c9d0d08fcd738edb5b

PR #4334 (2024-10-09): patch: remove gconf2 libgconf-2-4 deps
Closes #4096

Cherry-picked from https://github.com/balena-io/etcher/pull/4226

Thanks to @marcaurele 🙏
commit 2ed779ef (2024-10-09): patch: remove gconf2 libgconf-2-4 deps

Closes #4096