balena-io/etcher #4098: Failed to flash anything under void linux

setarcos, 2023-08-11:
- **Etcher version:**   1.8.11
- **Operating system and architecture:**  Void linux x86 (glibc)
- **What happened:**
When "Flash" was pressed, only following message was printed, and the UI returned to select target.
[18030:0811/172456.730042:ERROR:bus.cc(398)] Failed to connect to the bus: Failed to connect to socket /run/dbus/system_bus_socket: No such file or directory
[18030:0811/172456.730107:ERROR:bus.cc(398)] Failed to connect to the bus: Failed to connect to socket /run/dbus/system_bus_socket: No such file or directory
[18030:0811/172456.736056:ERROR:object_proxy.cc(623)] Failed to call method: org.freedesktop.DBus.Properties.Get: object_path= /org/freedesktop/portal/desktop: org.freedesktop.DBus.Error.InvalidArgs: No such interface “org.freedesktop.portal.FileChooser”
[18030:0811/172456.736087:ERROR:select_file_dialog_linux_portal.cc(242)] Failed to read portal version property
Language changed to: en
packageUpdatable false
ready-to-show: 752.228ms
Error occurred in handler for 'disable-screensaver': No handler registered for 'disable-screensaver'
Error occurred in handler for 'enable-screensaver': No handler registered for 'enable-screensaver'
- **Do you see any meaningful error information in the DevTools?** Nothing special besides the console messages.

setarcos, 2023-08-28:
I missed out a "user did not grant permission." message from devtools.  So it works with sudo.

closed 2023-08-28 by setarcos

lurch, 2023-08-28:
Possibly related to polkit? https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#no-polkit-authentication-agent-found-error-in-gnulinux
:shrug: