balena-io/etcher #4103: Release hashsums for AppImage in "latest-linux.yml" gone

TwizzyDizzy, 2023-09-01:
- **Etcher version:** not relevant for this issue
- **Operating system and architecture:** not relevant for this issue
- **Image flashed:** not relevant for this issue
- **What do you think should have happened:** the release artifact `latest-linux.yml` should have containted the base64 encoded sha512sum of the `.AppImage` release artifact
- **What happened:** Up until release `1.18.8`, the `latest-linux.yml` release artifact contained the base64 encoded sha512sum of the `.AppImage` release artifact, but from version `1.18.9` on, the hashsum for the `.AppImage` does not seem to be released as artifact anymore. The `latest-linux.yml` now contains the base64 encoded sha512sum of the 64 bit debian package.
- **Do you see any meaningful error information in the DevTools?** not relevant

Up until `1.18.8` - `latest-linux.yml`:
```
version: 1.18.8
files:
  - url: balenaEtcher-1.18.8-x64.AppImage
    sha512: oy0igwNmAjuv3Fb74KySK2scsQYMOWqg1QoQOqe9Qr9r+/HoGBoAIQ0DCUu4cj5KcnWqhC0emoXQvOVC64Kt3A==
    size: 99670674
    blockMapSize: 105126
path: balenaEtcher-1.18.8-x64.AppImage
sha512: oy0igwNmAjuv3Fb74KySK2scsQYMOWqg1QoQOqe9Qr9r+/HoGBoAIQ0DCUu4cj5KcnWqhC0emoXQvOVC64Kt3A==
releaseDate: '2023-04-26T09:42:25.874Z'
stagingPercentage: "100"
```

From `1.18.9` - `latest-linux.yml`:
```
version: 1.18.9
files:
  - url: balena-etcher_1.18.9_amd64.deb
    sha512: rF9LPcOlqgOSD8lc99Gj/BgMZuPRxCa558/6K8sNebqLQiHoBfWy12uGoUnX8AJ0m/+XOmfGssFs9cyaxFdPMg==
    size: 90103074
path: balena-etcher_1.18.9_amd64.deb
sha512: rF9LPcOlqgOSD8lc99Gj/BgMZuPRxCa558/6K8sNebqLQiHoBfWy12uGoUnX8AJ0m/+XOmfGssFs9cyaxFdPMg==
releaseDate: '2023-07-12T08:53:38.340Z'
stagingPercentage: "100"
```

I can haz hashsums for `.AppImage` again? 🥺

Cheers
Thomas

lurch, 2023-09-01:
Hmmm, as you point out in [release 1.18.8](https://github.com/balena-io/etcher/releases/tag/v1.18.8) the `latest-linux.yml` contains the checksum for (only) the `.AppImage` file, but in [release 1.18.11](https://github.com/balena-io/etcher/releases/tag/v1.18.11) the `latest-linux.yml` contains the checksum for (only) the `.deb` file.
These both feel equally arbitrary - surely the `latest-linux.yml` should contain the checksums for the `.AppImage`, the `.deb` and the `.rpm` files?

Similarly, the `latest.yml` only contains the checksum for `balenaEtcher-Setup-1.18.11.exe` whereas I guess it "should" include `balenaEtcher-1.18.11-ia32-win.zip`, `balenaEtcher-1.18.11-win.zip`, `balenaEtcher-Portable-1.18.11.exe` and `balenaEtcher-Setup-1.18.11.exe` ? :thinking:

TwizzyDizzy, 2023-09-01:
I do agree with all of your above points in that it feels kinda arbitrary, which hash sum is in which `.yml`.

If it were up to me, I'd simply either create a `.yml` file for every artifact or have one `all-in-one-hashsums.yml` which contains the sum for every artifact.

Cheers
Thomas

PR #4132 (2023-12-21): Modernize build pipeline
Switch to electron-forge which is now officially adopted by the Electron project. Solves a bunch of issues along the way.

Issues related to experimental v1.18.13:

Closes #4138 
Closes #4129
Closes #4119 

Build related issues:

Closes #4136 
Closes #4125 
Closes #4103 
Closes #4087 
Closes #3767 
Closes #3728 
Closes #3653 
Closes #3400 
Closes #3292 
Closes #3269 
Closes #3263 
Closes #2870 
Closes #2826 

(Admittedly, some of the issues are linked opportunistically -- please reopen them if the issue is not actually solved)
closed 2023-12-21 by dfunckt