balena-io/etcher #4107: Flashing USB drive for pi4B from ChromeBook

dlennard21, 2023-09-25:
- **Etcher version:** 1.18.11
- **Operating system and architecture:**  PixelBook, Chrome OS 116.0.5845.168,  Linux enabled
- **Image flashed:** chromiumos_image_r108-rpi4b.img.xz 
- **What do you think should have happened:** <

1. enabled Linux on Chromebook
2. downloaded and ran Balena Etcher 1.18.11
3. downloaded the Chromium image  to Downloads
4. Shared Downloads folder to Linux
5. Enabled Linux to see and manage my USB devices
6. Ran Etcher
7. Selected the Chromium image from **mbt/chromeos/MyFiles/Downloads**
8. Clicked Flash
9. and within 5 seconds a fail with the “Attention something went wrong..."dialogue as per screenshot>
10. and the USB drive disconnects at same time
11. - **What happened:** as above

**
[image]
[image]
[image]
[image]
[image]
[image]**
- **Do you see any meaningful error information in the DevTools?** No, but I wouldn't know where to look
-

lurch, 2023-09-25:
See https://github.com/balena-io/etcher/issues?q=is%3Aissue+chromebook and https://github.com/raspberrypi/rpi-imager/issues/92

From what I can tell (I don't have a Chromebook myself), ChromeOS only gives Linux "high level access" to the USB devices, and doesn't allow the "low level access" needed by tools like Etcher (needed to write to the disk at a block-level instead of the filesystem-level).

dlennard21, 2023-09-26:
Many thanks [lurch](https://github.com/lurch) for that. It seems pretty terminal then (excuse the pun).

Just to be clear : does _anyone_ have experience of flashing an SD card with chromium for a pi4B from a ChromeBook or PixelBook? Or as [lurch](https://github.com/lurch) says, is it impossible?