balena-io/etcher #4111: Request to Fix the Security Issue reported [Private]

febinrev, 2023-10-05:
Hi,

 This is Febin Mon Saji, an independent Security researcher. I reported a root privilege escalation via race-condition vulnerability (for linux) couple of months back, via security@balena.io and acknowledged by Balena team and even got a Swag pack. I hope you the developers are aware of the vulnerability and are already working on the patch and hope the newer version will be released soon.

As the vendor confirmed the issue and working on the patch, I wish to apply for a CVE id for the said vulnerability.

Let me know your thoughts.

Thanks.

aethernet, 2023-10-13:
As expressed privately (adding it here for transparency), we're in the middle of a quite deep revamp to address this and other issues.

Thanks for being patient, we take this seriously.

febinrev, 2023-11-30:
I hope the issue is fixed in the new pre-release version v1.18.13.