balena-io/etcher #4112: Unable to resize screen -- cannot see all devices

pushshift, 2023-10-07:
We are unable to resize the screen. When we have 10 USB devices connected at once, we are unable to see all devices in order to select them for mass creation of new boot devices. Cannot scroll down either to see other devices.

[image]

fibbbonachos, 2023-10-12:
Working on this 👍

fibbbonachos, 2023-10-12:
@pushshift Scroll seems to be working fine for me (users should be able to scroll down with the mouse wheel or a trackpad). Are you talking about adding a visual scrollbar?

lurch, 2023-10-13:
> users should be able to scroll down with the mouse wheel or a trackpad

Ah! You've just reminded me of #3395

pushshift, 2023-10-13:
@fibbbonachos -- I tried scrolling with the trackpad but didn't have much luck. I'll try again or try with a mouse that has a scroll wheel. Ideally I think a little visual scroll indicator would cover all setups / use cases. I'll update on Monday when I'm back to see if I can get it to work with the trackpad using my laptop.

pushshift, 2023-10-13:
@lurch Ahhh so this is actually just a copy of #3395. I noticed that is closed, though.

My vote would be to just add a cool looking scroll bar on the side when it overflows and that would handle everything. Up to you guys though -- If it isn't much work it would make the UI more functional. Either way, if using a scroll wheel on a mouse works, we'll make it work with that. 

Thanks for all your work on this product! Definitely is the nicest one I've used.

lurch, 2023-10-13:
> My vote would be to just add a cool looking scroll bar on the side when it overflows and that would handle everything.

FWIW that would also be my vote, but a) I'm not a UI / UX designer and b) I don't work for Balena any more :slightly_smiling_face: