balena-io/etcher #4113: read-only after flashing.

KenchanPrfm, 2023-10-08:
- **Etcher version:** 1.14.3 
- **Operating system and architecture:**Ubuntu 22.04.3 arm64
- **Image flashed:** Endeavouros_Cassini_Nova-03-2023_R3.iso
- **What do you think should have happened:** 
- **What happened:** Unable to recover USB stick from read-only file system.
- **Do you see any meaningful error information in the DevTools?** 

I read user document of etcher(https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#recovering-broken-drives), and I tried to recover by dd command.
But dd command has failed.
dd can not open `/dev/sdc`: read-only file system

Can I fix it ?

fibbbonachos, 2023-10-09:
Did you run it as `root`? as mentioned in the documentation? @KenchanPrfm

KenchanPrfm, 2023-10-10:
Yes, I have run both use with sudo and after "su" command.

I paste log bellow.
The Japanese prompt means "can not open '/dev/sdc' : read-only file system.".
```
root@xxx:/home/me# dd if=/dev/zero of=/dev/sdc bs=512 count=1 conv=notrunc
dd: '/dev/sdc' を開けませんでした: 読み込み専用ファイルシステムです
me@xxx:~$ sudo dd if=/dev/zero of=/dev/sdc bs=512 count=1 conv=notrunc
dd: '/dev/sdc' を開けませんでした: 読み込み専用ファイルシステムです
```

pushshift, 2023-10-12:
@KenchanPrfm Have you tried a different device? Are you sure the device you are using isn't experiencing a failure? How was the device mounted? Have you tried unmounting and remounting the device to see if it is still read-only? Also, check your syslog and see if there is any other additional information related to that device / drive. 

I've seen something like this happen with a failing thumb drive. If there is an issue, usually it will attempt to mount as read-only. I'd recommend running some diagnostics and checking syslog for any other information / clues.