balena-io/etcher #4129: Bug, Something went wrong...

i7i4, 2023-10-26:
Etcher version 1.18.13
Debian 12, kernel 6.1.0-13-amd64
Installed file: balena-etcher_1.18.13_amd64.deb

ctr+shift+i output:

gui.js:3 Successfully connected to IPC server: etcher-server-17434-1698307638907-unprivileged}}, socket root /run/user/1000/
gui.js:3 Object
gui.js:3 Error: Error invoking remote method 'disable-screensaver': Error: No handler registered for 'disable-screensaver'
    at a.invoke (node:electron/js2c/renderer_init:2:7723)
(anonymous) @ gui.js:3
gui.js:1  0 devices, 0% at 0.00 MB/s (total 0.00 MB/s)  with 0 failed devices
gui.js:3 Successfully connected to IPC server: etcher-server-17434-1698307648088-privileged}}, socket root /run/user/1000/
gui.js:3 params Object
gui.js:3 Image: /home/xyz/Downloads/xyz.iso
gui.js:3 Devices: /dev/sdc
gui.js:3 Auto blockmapping: false
gui.js:3 Decompress first: true
gui.js:3 fail event
gui.js:3 Object
gui.js:3 Object
gui.js:3 Flash results Object
gui.js:3 Object
gui.js:3 Terminating IPC server
gui.js:3 Error: The writer process ended unexpectedly
    at createError (gui.js:3:225919)
    at Object.createUserError (gui.js:3:226132)
    at finish (gui.js:307:2613)
    at Server.onFail (gui.js:307:2100)
    at Server.emit (gui.js:3:445008)
    at Server.gotData (gui.js:3:450365)
    at Socket.emit (node:events:513:28)
    at addChunk (node:internal/streams/readable:324:12)
    at readableAddChunk (node:internal/streams/readable:293:11)
    at Readable.push (node:internal/streams/readable:234:10)
    at Pipe.onStreamRead (node:internal/stream_base_commons:190:23)
(anonymous) @ gui.js:3
gui.js:3 Error: Error invoking remote method 'enable-screensaver': Error: No handler registered for 'enable-screensaver'
    at a.invoke (node:electron/js2c/renderer_init:2:7723)
(anonymous) @ gui.js:3
gui.js:1  0 devices, 0% at 0.00 MB/s (total 0.00 MB/s)  with 0 failed devices

linuxmaniac, 2023-10-28:
[image]

Same here, installed same deb version:
```
Linux debrock 6.1.0-13-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.55-1 (2023-09-29) x86_64 GNU/Linux
```

linuxmaniac, 2023-10-28:
```
dpkg: warning: downgrading balena-etcher from 1.18.13 to 1.18.11
```
And now works without an error

sgtcoder, 2023-10-29:
Yes, I downgraded to 1.18.1 (since it mentioned in another thread) and it fixed the problem as well.

PR #4132 (2023-12-21): Modernize build pipeline
Switch to electron-forge which is now officially adopted by the Electron project. Solves a bunch of issues along the way.

Issues related to experimental v1.18.13:

Closes #4138 
Closes #4129
Closes #4119 

Build related issues:

Closes #4136 
Closes #4125 
Closes #4103 
Closes #4087 
Closes #3767 
Closes #3728 
Closes #3653 
Closes #3400 
Closes #3292 
Closes #3269 
Closes #3263 
Closes #2870 
Closes #2826 

(Admittedly, some of the issues are linked opportunistically -- please reopen them if the issue is not actually solved)
closed 2023-12-21 by dfunckt