balena-io/etcher #4137: gconf dependencies no longer needed on Kubuntu 23.10

NotizMe, 2023-11-13:
- **Etcher version:** v1.18.12 (because v1.18.13 doesn't work on Kubuntu 23.10)
- **Operating system and architecture:**  Kubuntu 23.10 x64
- **Image flashed:** debian-12.2.0-i386-netinst.iso (https://cdimage.debian.org/debian-cd/current/i386/iso-cd/debian-12.2.0-i386-netinst.iso)
- **Reproduction process:**
```bash
git clone git@github.com:balena-io/etcher.git
cd etcher
# Using v1.18.12, because v1.18.13 is broken on Kubuntu 23.10
git checkout v1.18.12
# Getting .resinci.json see issue #4136 
git checkout 292f86d6f5b0e8dd34cb3dd6e008517f9a066cd0 .resinci.json
make electron-develop
make electron-build
sudo apt install ./dist/balena-etcher-electron_1.18.12+09e13e9b_amd64.deb
```
- **What do you think should have happened:** Etcher should have installed and I can flash my image
- **What happened:**
```
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'balena-etcher-electron' instead of './dist/balena-etcher-electron_1.18.12+09e13e9b_amd64.deb'
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 balena-etcher-electron : Depends: gconf2 but it is not installable
                          Depends: gconf-service but it is not installable
                          Depends: libgconf-2-4 but it is not installable
E: Unable to correct problems, you have held broken packages.
```
- **Do you see any meaningful error information in the DevTools?** N/A
# Extra info
If I just removed all gconf dependencies everything worked. I was able to install Etcher and flash my image.
```bash
sed -i '/gconf/d' electron-builder.yml
sed -i '/gconf/d' docs/USER-DOCUMENTATION.md
sed -i '/gconf/d' scripts/resin/electron/electron-builder.json
make electron-build
sudo apt install ./dist/balena-etcher-electron_1.18.12+09e13e9b_amd64.deb
```
If this is the solution this should solve:
- #4134
- #4096

I didn't make a Pull Request because I'm not familiar with the codebase and I have 2 other interfering issues that make it impossible to test my fix on the master branch:
- #4136
- (will be added when created)

theofficialgman, 2023-12-02:
yup gconf is no longer available in debian trixie and ubuntu 23.10+. gconf was deprecated by upstream gnome 8 years ago. it should have been removed a long time ago.

qwertychouskie, 2024-02-12:
@NotizMe Since the other two issues seem to be fixed, any chance you could create that PR?

thynus, 2025-02-26:
@aethernet this bug can be closed https://github.com/balena-io/etcher/commit/2ed779ef371db367e4e413c9d0d08fcd738edb5b