balena-io/etcher #4141: Etcher fails on Kali ISO: Failed to load resource: net::ERR_FILE_NOT_FOUND index.css:1 & No handler registered for 'enable-screensaver'

GrnEydGuy, 2023-11-21:
- **Etcher version:**     
- 1.8.11
- 
- **Operating system and architecture:**   
- Windows 10 64bit
- 
- **Image flashed:**   
- https://cdimage.kali.org/kali-2023.3/kali-linux-2023.3-live-amd64.iso
- 
- **What do you think should have happened:** 
- 
- 1. Insert Brand New PNY 128 gb USB 3.2 to flash Kali LIve image.  
- 2. open Etcher
- 3. select image (downloaded on my hard drive) also tried using image from url
- 4. select target USB
- 5. Flash - completes, then verifies then FAILS
- 
- **What happened:**
- Etcher reports FLASH FAILS
- windows pop up opens saying the disk in F: is unusable and needs to be formatted
- 
- **Do you see any meaningful error information in the DevTools?** 
- Failed to load resource: net::ERR_FILE_NOT_FOUND  index.css:1
- Error: Error invoking remote method 'enable-screensaver': No handler registered for 'enable-screensaver'
    at o.invoke (node:electron/js2c/renderer_init:57:526)   gui.js:37

- OTHER INFO:  I have tried running as admin, disabling windows defender, reinstalling etcher, validating hash files.  Same errors repeat every times regardless.

ArSo12, 2023-11-27:
I have the same problem with different image. Also tried admin, defender etc.

Wesley5n1p35, 2023-12-16:
Same Issue here. I dont think its a Balena issue as I downloaded Rufus just to troubleshoot and it failed as well.

GrnEydGuy, 2023-12-16:
> Same Issue here. I dont think its a Balena issue as I downloaded Rufus just to troubleshoot and it failed as well.
> 

Rufus works fine if using DD mode and then switching bios to legacy. I had no problems with Rufus. This is an Etcher issue.

Wesley5n1p35, 2023-12-16:
> > Same Issue here. I dont think its a Balena issue as I downloaded Rufus just to troubleshoot and it failed as well.
> 
> Rufus works fine if using DD mode and then switching bios to legacy. I had no problems with Rufus. This is an Etcher issue.

How do you do that?

GrnEydGuy, 2023-12-16:
> > > Same Issue here. I dont think its a Balena issue as I downloaded Rufus just to troubleshoot and it failed as well.
> > 
> > Rufus works fine if using DD mode and then switching bios to legacy. I had no problems with Rufus. This is an Etcher issue.
> 
> How do you do that?

You should see a pop up when you initiate Rufus that asks if you'd like to burn the image in DD mode. As far as setting you legacy boot it depends on your bios. My suggestion is to search the internet to find out how to enter the bios on your particular PC.