balena-io/etcher #4142: error: file `/boot/` not found  Before Live Grub Menu

usero1020, 2023-11-22:
- **Etcher version:**  balenaEtcher-1.18.11-x64.AppImage
- **Operating system and architecture:** Xubuntu 22.04
- **Image flashed:** Xubuntu 22.04
- **What do you think should have happened:**
It should have flashed with no negative after effects when booting the USB.
- **What happened:**
Balena completed successfully as well as the Validation.   The problem is, when i booted the USB on the PC in UEFI mode, just before grub menu shows up, it flashes quickly an error: 

> error: file `/boot/` not found

I can proceed to install or try Xubuntu, but this error that flashes before the grub menu on the live USB boot concerns me.

The reason it concerns me is because when i flash the same image with Rufus, this error does not appear when booting the live USB.

The error does nto prevent me from booting into Xubuntu and installing it, i just want to know why this happens and why when using Rufus there is no error like that that flashes before the live grub menu.

lurch, 2023-11-22:
Sounds like a question for the Xubuntu developers?

usero1020, 2023-11-22:
Why would it be a question for the Xubuntu developers if when flashing with Rufus there is no error?  This is a flashing issue.

lurch, 2023-11-22:
Because Rufus works differently to Etcher (which https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md also hints at).

Rufus (and UNetbootin) partition the USB drive and make it bootable, and _then_ extract the data from the ISO onto this bootable drive. But this requires Rufus to have internal knowledge about how the ISO is set up (and that's why there's a list of "supported ISOs" at the bottom of the [Rufus homepage](https://rufus.ie/en/), and why the Rufus changelog says things like "Fix persistence support for Debian 12" and "Improve support for Slax Linux".).

By contrast, Etcher (and Win32DiskImager and RaspberryPiImager) simply extract the raw ISO / disk-image directly onto the USB drive. This means that they work with _any_ disk image (without needing to be updated as new OSes get released), and not just ISO images intended for x86 PCs. Whatever is in the disk image, is exactly byte-for-byte what you end up with on your USB stick.

This is why you [can't use Etcher to make a bootable Windows USB drive](https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#why-is-my-drive-not-bootable) - because the Windows ISO is setup to be burnt to a DVD disc, and not directly written to a USB drive. However Rufus has internal knowledge about the Windows ISO, and knows how to convert it into a bootable USB drive.

And then things like the Ubuntu live images are "hybrid ISOs" which can either be burnt to a DVD disc **or** written directly to a USB drive - but this sometimes comes with it's [own problems](https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#flashing-ubuntu-isos).

---

So in summary: because Etcher doesn't do any modifications to the disk-image, but just raw-copies the disk-image onto the USB drive, any errors that you get while booting must have been contained in the original image, rather than be anything done by Etcher.

usero1020, 2023-11-23:
@lurch Thank you for the explanation.

I did another test with Rufus now and flashed the drive in dd mode (instead of in ISO mode) and it gave the same error as Balena.  Which leads me to believe that Balena by default flashes the ISO in dd mode as well, correct?
Im just not sure if this happens with Ubuntu Desktop as well (i havent tried it).  If it doesnt, then that means indeed it is a question for the Xubuntu developers.

lurch, 2023-11-23:
> Which leads me to believe that Balena by default flashes the ISO in dd mode as well, correct?

Yes, "dd mode" is the same as the "exactly byte-for-byte" that I mentioned in my comment above. I've now edited my comment to mention `dd` too.