balena-io/etcher #4144: Cannot install on Ubuntu 23.10; unmet dependency gconf2

MorphZG, 2023-12-06:
I am running Ubuntu 23.10. Trying to install Balena Etcher, downloaded few latest .deb releases but also tried with 1.17.0. Tried to install downloaded package with:
```
sudo apt install ./balena-etcher_1.17.0_amd64.deb
```
Than someone proposed to add the repository with command below and than try to install

```
curl -1sLf  'https://dl.cloudsmith.io/public/balena/etcher/setup.deb.sh'  |  sudo -E bash
```

In the end i always get the following error, saying there is unmet dependency `gconf2`. But it is unavailable                         
```
The following packages have unmet dependencies:
 balena-etcher : Depends: gconf-service but it is not installable
                 Depends: gconf2 but it is not installable
                 Depends: libgconf-2-4 but it is not installable
```

After trying to install gconf2:
```
Package 'gconf2' has no installation candidate
```

MorphZG, 2023-12-06:
> Try `$sudo apt --fix-broken install`

Nothing to fix since nothing was installed.

lurch, 2023-12-06:
Duplicate of #4096 ?

> Than someone proposed to add the repository

See #4059

scorninpc, 2023-12-15:
Same on debian

Download the deb package here and install 

https://packages.debian.org/bookworm/amd64/gconf2/download

aryzing, 2024-05-14:
dupe https://github.com/balena-io/etcher/issues/4096

sam-txts, 2024-06-14:
> I am using Kali Linux. Since it's a debian based distro, I faced the same problem as you while trying to install the .deb file downloaded from the balena-io etcher's releases page.
> 
> The App Image file didn't work for me either. So I tried to look for a way around.
> 
> I have managed to find a solution. Instead of downloading the .deb package, download the .rpm package and convert into .deb using `alien`.
> 
> Install the `alien` package using the following command:
> 
> ```
> sudo apt update
> sudo apt install alien
> ```
> 
> Now navigate to the directory you have the .rpm package. And run the following command:
> 
> ```
> sudo alien balena-etcher-1.19.21-1.x86_64.rpm
> ```
> 
> It will take a while. Now install the .deb package that has been generated.
> 
> ```
> sudo dpkg -i balena-etcher_1.19.21-2_amd64.deb 
> ```
> 
> Et voila.
> 
> ![image](https://private-user-images.githubusercontent.com/86034982/337975987-e6a32ba1-319e-4aa0-a28f-d9216444cd10.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTgzODUxNDUsIm5iZiI6MTcxODM4NDg0NSwicGF0aCI6Ii84NjAzNDk4Mi8zMzc5NzU5ODctZTZhMzJiYTEtMzE5ZS00YWEwLWEyOGYtZDkyMTY0NDRjZDEwLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MTQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjE0VDE3MDcyNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTkyYzJkYWM3N2RjMjM1ZWQ5YzQxMDA5ZDA5NWY4ZjJmNjEzNDNhNTk4MzgwNDQzNzY2ZmY3NDU4OGM1ODI1MGImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.jg_lF2TQ0pMhda20ZeIC-9Wf24sihTGmTjzKF3a6bfg)
> 
> Finally!
> 
> ![image](https://private-user-images.githubusercontent.com/86034982/337975915-3f69dc01-7c6a-4d31-b7c7-40cfbce26d8b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTgzODUxNDUsIm5iZiI6MTcxODM4NDg0NSwicGF0aCI6Ii84NjAzNDk4Mi8zMzc5NzU5MTUtM2Y2OWRjMDEtN2M2YS00ZDMxLWI3YzctNDBjZmJjZTI2ZDhiLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MTQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjE0VDE3MDcyNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWIzOWJhNjMxZmEwNjBjODIyMGM1NjM5MTJhODk4YzFlYTQ5NTM4NmJlYTg2YTQ1ZDI2NDU3NTJkZTc5ZTlkMGQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.XAlbMa6O8HF5dfKbPCdieLm4j_XZUwzlu57JYgo0yko)

well for some reason it won't work for me
[image]

sam-txts, 2024-06-14:
well for some reason it wont work for me 
[image]

AgSpades, 2024-06-14:
> well for some reason it wont work for me 
> [image]
> 

Did you check whether the .rpm file is present inside the directory you are working on? If not, download .rpm file from the [releases](https://github.com/balena-io/etcher/releases/) page and try again.

sam-txts, 2024-06-14:
it was in the same directory. i even reinstalled the rpm but still it shows the same error. i use kubuntu 24.04

sam-txts, 2024-06-14:
> it was in the same directory. i even reinstalled the rpm but still it shows the same error. i use kubuntu 24.04

might i try reinstalling that alien thingy how do i do so im new to this

aryzing, 2024-06-14:
@MorphZG seems the proposal above is more of a workaround than a solution?

MorphZG, 2024-06-15:
@aryzing should i edit the title once again?

lurch, 2024-06-15:
@sam-txts WRT `dl.bintray.com/etcher` and `deb.etcher.io` see #4059

t-bosss, 2024-06-22:
In my case it gets open and closed at the same time

t-bosss, 2024-06-22:
I'm trying Ubuntu 24.04 LTS and get the same error to wireguard-gui

shourya28, 2024-06-24:
> SIMPLE ANSWER! run it as `sudo apt --fix-broken install ./balena-etcher(version id here).deb` but make sure you have the bookworm repo in your sources beforehand! :p

This worked for me. Thanks!

TapTSEC-dev-2, 2024-07-23:
> I'm trying Ubuntu 24.04 LTS and get the same error to wireguard-gui

try getting ProtonVPN, it has WireGuard built in

TapTSEC-dev-2, 2024-07-23:
> > SIMPLE ANSWER! run it as `sudo apt --fix-broken install ./balena-etcher(version id here).deb` but make sure you have the bookworm repo in your sources beforehand! :p
> 
> This worked for me. Thanks!

no problem! im so glad i could help :p

bas-kirill, 2024-08-09:
```
wget http://ftp.de.debian.org/debian/pool/main/g/gconf/gconf2-common_3.2.6-8_all.deb && wget http://ftp.de.debian.org/debian/pool/main/g/gconf/libgconf-2-4_3.2.6-8_amd64.deb && wget http://ftp.de.debian.org/debian/pool/main/g/gconf/gconf2_3.2.6-8_amd64.deb && wget http://ftp.de.debian.org/debian/pool/main/g/gconf/gconf-service_3.2.6-8_amd64.deb && sudo dpkg -i gconf2-common_3.2.6-8_all.deb && sudo dpkg -i libgconf-2-4_3.2.6-8_amd64.deb && sudo dpkg -i gconf-service_3.2.6-8_amd64.deb && sudo dpkg -i gconf2_3.2.6-8_amd64.deb && rm -rf *conf*.deb
```

And installing deb from there:
```
  http://security.ubuntu.com/ubuntu/pool/main/o/openldap/libldap-2.5-0_2.5.16+dfsg-0ubuntu0.22.04.2_amd64.deb
```

solved the issue

---

before:
[image]

after:
[image]

YASIRUXYZ, 2024-09-17:
# Install balenaEtcher in debian systems (ubuntu)

### create temporary folder for install dependencies
```
sudo su
cd /tmp/ &&
mkdir /tmp/balenaServiceTmp &&
cd /tmp/balenaServiceTmp
```
### download dependencies
```
wget http://security.ubuntu.com/ubuntu/pool/main/o/openldap/libldap-2.5-0_2.5.16+dfsg-0ubuntu0.22.04.2_amd64.deb >> libldap.deb
wget http://ftp.de.debian.org/debian/pool/main/g/gconf/gconf2-common_3.2.6-8_all.deb 
wget http://ftp.de.debian.org/debian/pool/main/g/gconf/libgconf-2-4_3.2.6-8_amd64.deb
wget http://ftp.de.debian.org/debian/pool/main/g/gconf/gconf2_3.2.6-8_amd64.deb 
wget http://ftp.de.debian.org/debian/pool/main/g/gconf/gconf-service_3.2.6-8_amd64.deb 
```

### install dependencies
```
sudo dpkg -1 libldap.deb
sudo dpkg -i gconf2-common_3.2.6-8_all.deb 
sudo dpkg -i libgconf-2-4_3.2.6-8_amd64.deb 
sudo dpkg -i gconf-service_3.2.6-8_amd64.deb
sudo dpkg -i gconf2_3.2.6-8_amd64.deb 
```

### download & install balenaEtcher
```
wget https://github.com/balena-io/etcher/releases/download/v1.19.22/balena-etcher_1.19.22_amd64.deb >> balenaEtcher.deb
sudo apt install ./balenaEtcher.deb
```
### remove temporary folder
```
sudo rm -rf /tmp/balenaServiceTmp
```

### run balenaEtcher
```
balena-etcher
```

MichaelLFR, 2024-10-05:
Thanks !
`wget http://security.ubuntu.com/ubuntu/pool/main/o/openldap/libldap-2.5-0_2.5.16+dfsg-0ubuntu0.22.04.2_amd64.deb >> libldap.deb`
Causes "unexpected end of file". Better downloading and renaming after. Or keeping the original file name.
`sudo dpkg -1 libldap.deb`
-i to replace -1 :)