balena-io/etcher #4150: Error: (0 , h.requestMetadata) is not a function

wegank, 2023-12-21:
- **Etcher version:** 1.19.0
- **Operating system and architecture:** NixOS, x86_64-linux
- **Image flashed:** Arbitrary image
- **What do you think should have happened:** Image selected successfully, being able to select target
- **What happened:** Can't select any image
- **Do you see any meaningful error information in the DevTools?**  No, same error as above.

[image]

In Nixpkgs we package Etcher on the basis of the official deb. The procedure no longer works as of 1.18.13.

Likely a continuation of #4138, but not tested on other distros.

Sargeanthost, 2023-12-24:
Also having this issue on Manjaro

devM7MD, 2023-12-24:
Like you but I'm using ArchLinux :(

polsvoice, 2023-12-25:
I'm getting this with version 1.18.14 on Garuda.

myxor, 2023-12-26:
Same for me on ArchLinux.

OffsetMonkey538, 2023-12-26:
Getting this on Nobara 38

Sargeanthost, 2023-12-26:
Was initially trying on Hyprland, which is on Wayland. I download Cinnamon to test and Xorg environment and it worked for me. I used 1.8.13. Is anyone else on Wayland?

OffsetMonkey538, 2023-12-26:
I was on wayland

TerrorBite, 2024-01-01:
I'm getting this issue on Manjaro with the 1.18.14 version from the AUR and I'm on X11. However, it lools like 1.18.14 released just a couple of days before #4132 was merged, meaning that the fix for #4138 wasn't included in that version. So any Arch/Manjaro issues (@Sargeanthost, @devM7MD, @myxor) should hopefully be fixed when 1.19.x lands on the AUR. Maybe I can try and manually build a 1.19 package and see if it works?

TerrorBite, 2024-01-01:
It looks like the move to building with electron-forge causes build issues on Arch, and this is why the AUR package is still on version 1.18.

TobiPeterG, 2024-01-04:
I have the same issue on openSUSE using the RPM and appimage

LITUATUI, 2024-01-08:
Same problem here with EndeavourOS.

slagathor69, 2024-01-14:
Same issue on Ubuntu 22.04 with X11

shunkica, 2024-01-16:
Same issue with balena-etcher-1.19.3-1.x86_64.rpm on Fedora 39

eternallfrost, 2024-01-17:
This issue does not occur on the appimage version of Etcher

dualrelax, 2024-01-18:
I use archlinux ,balenaEtcher-1.18.8-x64.AppImage can use

aethernet, 2024-01-23:
Thanks for the reports guys.
This basically means the sub-process `etcher-utils` could not be spawned properly.
There might be more infos in the debug console. 
Can you check what it says ( Ctrl + Shift + I should open it) ?

ClassANetwork, 2024-01-23:
> Thanks for the reports guys. This basically means the sub-process `etcher-utils` could not be spawned properly (or is not yet ready). There might be more infos in the debug console. Can you check what it says ( Ctrl + Shift + I should open it) ?

Getting the same on Fedora 39 with version v1.19.3. Tried both RPM and AppImage formats. Same result. 
[-1706048290684.log](https://github.com/balena-io/etcher/files/14030356/-1706048290684.log)

Kaiddd, 2024-02-03:
Same on Arch on both x11 and wayland, installed with yay from aur (since people above were talking abt wayland)

oscarcoghlan, 2024-02-09:
> Thanks for the reports guys. This basically means the sub-process `etcher-utils` could not be spawned properly (or is not yet ready). There might be more infos in the debug console. Can you check what it says ( Ctrl + Shift + I should open it) ?

```
node:events:491 Uncaught Error: spawn /usr/lib/balena-etcher/generated/etcher-util ENOENT
    at __node_internal_captureLargerStackTrace (node:internal/errors:490:5)
    at __node_internal_errnoException (node:internal/errors:620:12)
    at ChildProcess._handle.onexit (node:internal/child_process:283:19)
    at onErrorNT (node:internal/child_process:476:16)
    at process.processTicksAndRejections (node:internal/process/task_queues:82:21)
(anonymous) @ gui.js:3
gui.js:3 TypeError: requestMetadata is not a function
    at gui.js:278:1759
    at SourceSelector.selectSource (gui.js:278:2719)
    at SourceSelector.openImageSelector (gui.js:278:3117)
```

Unfortunately console does not have much info

JBlackCat, 2024-02-12:
OS: Fedora 39
pkg: balena-etcher-1.19.4-1.x86_64.rpm

Looking at the inspector the error is thrown in the `source-selector` component here... https://github.com/balena-io/etcher/blob/v1.19.4/lib/gui/app/components/source-selector/source-selector.tsx#L425

`requestMetadata` which is imported from `app` is still `undefined`.

https://github.com/balena-io/etcher/blob/v1.19.4/lib/gui/app/app.ts#L139-L157

Since `requestMetadata` is initially defined as `undefined` there needs to be an additional check in `selectSource` to know that `requestMetadata` has been defined. It appears the only way to know this is listening for the emitted `scan` event, but I didn't deep dive on a solution.

nath1as, 2024-02-12:
still an issue on arch

kdh8219, 2024-02-14:
Archlinux, Is this same problem?
[image]

harperbolic, 2024-02-16:
same problem, archlinux

Niskeletor, 2024-02-16:
Experiencing the same issue here on Manjaro and Garuda Linux with balena-etcher-2:1.18.14-1.

Just as @Kaiddd commented, I can confirm that the balenaEtcher-1.5.109-x64.AppImage version works fine on both systems.

wasantosfi, 2024-02-18:
Same issue here on ElementaryOS and Zorin OS with balenaEtcher-1.19.5-x64.AppImage.
Confirmed that balenaEtcher-1.18.11-x64.AppImage worked on both systems.

dmophir, 2024-02-19:
Experiencing this problem on my Dell XPS 9520 running Arch Linux. This is with balenaEtcher v1.18.14 from AUR, v1.18.12 AppImage, and v1.19.5 AppImage. Running GNOME 45.4 on Wayland with Intel GPU.

binarytrails, 2024-02-22:
@aethernet thanks for the ctrl+shift+l trick!
```
Failed to load resource: net::ERR_FILE_NOT_FOUND
gui.js:320  _____ _       _
|  ___| |     | |
| |__ | |_ ___| |__   ___ _ __
|  __|| __/ __| '_ \ / _ \ '__|
| |___| || (__| | | |  __/ |
\____/ \__\___|_| |_|\___|_|

Interested in joining the Etcher team?
Drop us a line at join+etcher@balena.io

Version = 1.18.14, Type = local
gui.js:3 [Intervention] Slow network is detected. See https://www.chromestatus.com/feature/5636954674692096 for more details. Fallback font will be used while loading: file:///usr/lib/balena-etcher/generated/lib/gui/app/css/fonts/SourceSansPro-SemiBold.ttf
node:events:491 Uncaught Error: spawn /usr/lib/balena-etcher/generated/etcher-util ENOENT
    at __node_internal_captureLargerStackTrace (node:internal/errors:490:5)
    at __node_internal_errnoException (node:internal/errors:620:12)
    at ChildProcess._handle.onexit (node:internal/child_process:283:19)
    at onErrorNT (node:internal/child_process:476:16)
    at process.processTicksAndRejections (node:internal/process/task_queues:82:21)
gui.js:3 Electron Security Warning (Insecure Content-Security-Policy) This renderer process has either no Content Security
  Policy set or a policy with "unsafe-eval" enabled. This exposes users of
  this app to unnecessary security risks.

For more information and help, consult
https://electronjs.org/docs/tutorial/security.
This warning will not show up
once the app is packaged.
(anonymous) @ gui.js:3
gui.js:3 TypeError: requestMetadata is not a function
    at gui.js:278:1759
    at SourceSelector.selectSource (gui.js:278:2719)
    at SourceSelector.openImageSelector (gui.js:278:3117)
```

arbaes, 2024-02-23:
Hello, same error but different traceback:

```js
index.css:1     Failed to load resource: net::ERR_FILE_NOT_FOUND
gui.js:320  _____ _       _
|  ___| |     | |
| |__ | |_ ___| |__   ___ _ __
|  __|| __/ __| '_ \ / _ \ '__|
| |___| || (__| | | |  __/ |
\____/ \__\___|_| |_|\___|_|

Interested in joining the Etcher team?
Drop us a line at join+etcher@balena.io

Version = 1.18.14, Type = local
node:events:491 Uncaught Error: spawn /usr/lib/balena-etcher/generated/etcher-util ENOENT
    at __node_internal_captureLargerStackTrace (node:internal/errors:490:5)
    at __node_internal_errnoException (node:internal/errors:620:12)
    at ChildProcess._handle.onexit (node:internal/child_process:283:19)
    at onErrorNT (node:internal/child_process:476:16)
    at process.processTicksAndRejections (node:internal/process/task_queues:82:21)
gui.js:3 Electron Security Warning (Insecure Content-Security-Policy) This renderer process has either no Content Security
  Policy set or a policy with "unsafe-eval" enabled. This exposes users of
  this app to unnecessary security risks.

For more information and help, consult
https://electronjs.org/docs/tutorial/security.
This warning will not show up
once the app is packaged.
(anonymous) @ gui.js:3
gui.js:3 TypeError: requestMetadata is not a function
    at gui.js:278:1759
    at SourceSelector.selectSource (gui.js:278:2719)
    at SourceSelector.openImageSelector (gui.js:278:3117)
(anonymous) @ gui.js:3
gui.js:3 TypeError: requestMetadata is not a function
    at gui.js:278:1759
    at SourceSelector.selectSource (gui.js:278:2719)
    at SourceSelector.openImageSelector (gui.js:278:3117)
(anonymous) @ gui.js:3

```

pamo22, 2024-02-26:
On arch linux, installed https://aur.archlinux.org/packages/etcher-bin from the AUR and it works, https://aur.archlinux.org/packages/balena-etcher shows the error. For what is worth, the etcher-bin pkg was last updated 2023-07-15, while balena-etcher was updated 2023-12-20.

binarytrails, 2024-02-26:
@pamo22 https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=etcher-bin looks legit and verifies binary sums, thanks!

Alex313031, 2024-03-07:
@aethernet @polsvoice @TerrorBite @myxor @JBlackCat @wegank @aethernet I have a fork I'm building, with Windows 7 support. I am also having this issue on MacOS and Windows (but not Linux, actually)

The thing is that `pkg` is not correctly bundling everything in `etcher-util`. Building 1.19.5 leads to a binary that is identical to the releases provided, except etcher-util is about half the size it should be. In the dev console, the same errors relating to etcher-util that people have noted above show up.

Integrating @aethernet your patch to use the new maintained pkg and node 20 types yields nothing. I also tried node 16, node 18, and node 20, and MSVS 2019 and MSVS 2022, to no avail. Also tried building it in git bash and MSYS2, again with no difference.

Alex313031, 2024-03-17:
@aethernet Pinging you again to take a look, as you seem to be the highest contributor, and this is affecting multiple users trying to run and trying to build Etcher on Linux & Windows (including me)

ShalokShalom, 2024-03-21:
This just caused NixOS to remove the application from the repositories.

See: https://github.com/NixOS/nixpkgs/pull/295853

GarlandKey, 2024-03-30:
**OS:** Arch Linux x86_64
**Host:** Framework 16 (AMD Ryzen 7040 Series) AJ 
**Kernel:** 6.8.1-arch1-1 

**Version:** balenaEtcher (1.19.5).AppImage

**Console log:**

```log
index.js:2 /tmp/.mount_balenaScyPgz/usr/lib/balena-etcher/resources/etcher-util
index.js:2 TypeError: (0 , h.requestMetadata) is not a function
    at index.js:140:1418
    at H.selectSource (index.js:140:2212)
    at H.openImageSelector (index.js:140:2543)
(anonymous) @ index.js:2
```

Alex313031, 2024-04-09:
@aethernet @dfunckt @klutchell Is someone going to look at this? I've been waiting 2 months now. It prevent making working builds on Linux and Windows. I have a fork that still supports Windows 7, and this is blocking me from doing any more work on it. pkg or yao-pkg is not bundling all the native modules into etcher-util properly. I say yao-pkg because I tried @aethernet two patches related to this (and updating to Node 20), with no luck.

ShalokShalom, 2024-04-09:
Development on Etcher seems to have stopped. Personally, I switched to Ventoy and do not look back at all.

Alex313031, 2024-04-10:
@ShalokShalom I might base my fork on the 1.18.x branch instead.

Dagwal, 2024-04-10:
> On arch linux, installed https://aur.archlinux.org/packages/etcher-bin from the AUR and it works, https://aur.archlinux.org/packages/balena-etcher shows the error. For what is worth, the etcher-bin pkg was last updated 2023-07-15, while balena-etcher was updated 2023-12-20.

Thanks!!! etcher-bin is worked for me

gaby64, 2024-04-14:
wtf, this is still not fixed?

lh1207, 2024-04-14:
this has been an issue for 5 months

lekonai, 2024-04-15:
thanks @lh1207 !

xZero707, 2024-04-15:
Sadly this remains unfixed. Etcher used to be a great tool.

florian6973, 2024-04-24:
Yes have it on Redhat

aethernet, 2024-04-25:
Guys, you're using pre-releases that are not marked as production ready for this exact reason.

6jarjar6, 2024-04-26:
Having this issue now on MacOS

djmadeira, 2024-04-26:
> Guys, you're using pre-releases that are not marked as production ready for this exact reason.

The problem for me at least is that the version available on AUR is an unstable release. Because Arch releases on a rolling basis there's no way for me to peg to the latest stable release other than checking out the source and building myself. So I think the issue is pushing prerelease builds to package repositories.

Alex313031, 2024-04-26:
@polsvoice @TerrorBite @myxor @gaby64 @wegank @lh1207 @GarlandKey @ShalokShalom @pamo22 @arbaes Looks like they finally fixed it with this commit > https://github.com/balena-io/etcher/commit/2dfa795129e287f887b9ea02f2eca717575d27ac

LeeBinder, 2024-04-27:
@Alex313031 that's great news, thanks for letting us all know 👍 

I can confirm this bug indeed fixed with https://github.com/balena-io/etcher/releases/tag/v1.19.16 🥇

ShalokShalom, 2024-04-28:
Can we close this? @wegank

RavenEibu, 2024-04-29:
I still having problems with this error in 1.19.16 i'm running Fedora 40

[image]

lh1207, 2024-05-06:
> I have the same problem on the Hyprland on archlinux i use the appimage.

Use yay instead
https://github.com/balena-io/etcher/issues/4150#issuecomment-2054177599

gaby64, 2024-05-06:
problem still present in AppImage, Ubuntu

wegank, 2024-05-06:
> Can we close this? @wegank

I can't test because of https://github.com/balena-io/etcher/issues/4150#issuecomment-2012436349, but it seems that the same issue persists for AppImage?

hising, 2024-05-06:
> this has been an issue for 5 months with the balena-etcher aur. upon removing balena-etcher and replacing with etcher-bin, the issue is resolved.
> 
> ```shell
> yay -R balena-etcher
> yay -S etcher-bin
> ```

This solved it for me on Manjaro.

ivanjx, 2024-05-07:
issue still exists on `balenaEtcher-1.19.16-x64.AppImage`
[image]

musicfunfan, 2024-05-07:
@lh1207 So , i install the appimage available in official etcher site and now is working. The problem is in ``1.19.16``. The version in the official site is ``1.18.11``. I think the latest stable version is ``1.18.11``. So this version we should use.

supercoderlee, 2024-05-12:
OS: Windows11 23H2(AMD Ryzen 5600G)
Version: balenaEtcher 1.19.16 and 1.19.17

The ISO image cannot be loaded after selecting it, and destination disk selection is not possible, but version 1.18.11 does not have this problem.

musicfunfan, 2024-05-13:
@supercoderlee I thing that the ``1.18.11``, Is the stable one. This is the one that exist on the site. I think the ``1.19*`` is on Beta or something like that.

M0rtale, 2024-05-20:
Still experiencing this issue on kubuntu 24.04, 1.18.11 is not available bc 24.04 ships with glibc-2.39
[image]

musicfunfan, 2024-05-21:
@M0rtale Weird, i am on archlinux i do not know what version of glibc i have... ( i am not on home machine now) but... ``1.18.11`` works fine. 

This is not a fix, but you can install [distrobox](https://github.com/89luca89/distrobox) and use a distro that balena etcher ``1.18.11`` works. Just do to your job...

[191 more replies]