balena-io/etcher #4151: How to open the software?

balz666, 2023-12-21:
I downloaded Etcher, using the git clone command... How do I open it? Sorry, I am a newbie, but I don't find any way to open the software.

lurch, 2023-12-22:
"git clone" is for if you want to build the software from source-code. If you just want to download and run it, see https://etcher.balena.io/#download-etcher

balz666, 2023-12-24:
I can't open appimage for some reason. I click "execute" and nothing happens.

Once I cloned it, what should I type to open the software?

roadapathy, 2023-12-26:
Download the balenaEtcher-1.18.11-x64.AppImage file (or whatever version it is when you read this)
chmod +x balenaEtcher-1.18.11-x64.AppImage
./balenaEtcher-1.18.11-x64.AppImage
And you should be good to go. The GIT is for downloaded the source code, compiling it, installing, then running it which you shouldn't need to do for this.