balena-io/etcher #4156: Balena Etcher is taking a long time to start

gogo100people, 2023-12-29:
- 1.18.11
- Tails Linux
- It should 
- I Waited 3 Hours And it didn't start

I Am Not Root I Am Using Windows

sipiguy, 2024-01-07:
I am having this same problem, new download (1/7/2024) Balena Etcher 1.18.11 on Win7, It puts up an empty balenaEtcher window and then hangs. Is it waiting for something?

sipiguy, 2024-01-07:
Same result running BalenaEtcher 1.18.11 Portable in user and administrator modes. OS is Win7 Home Premium 64-bit with service pack 1 and 8GB ram. Yes, I know this OS is dead, but the system is too slow to run Win10.  :(

aethernet, 2024-01-23:
win7 is not supported anymore, you can use an older version of etcher (1.17.9 should work IIRC)

> I Am Not Root I Am Using Windows 

Don't understand what you mean

gogo100people, 2024-01-23:
> win7 is not supported anymore, you can use an older version of etcher (1.17.9 should work IIRC)
> 
> > I Am Not Root I Am Using Windows
> 
> Don't understand what you mean

I dont know how root on windows works?

victorel-petrovich, 2025-06-10:
Windows 10, takes >= 1min30 seconds to start