balena-io/etcher #4161: Flash Failed reported on successful flash due to screensaver error

phillipremaker, 2024-01-08:
- **Etcher version:** 1.8.11
- **Operating system and architecture:** Windows 11 x64
- **Image flashed:** raspbx-04-04-2018.img (and other raspbx images)
- **What do you think should have happened:** No error report on successful flash
- **What happened:** Error reported on successful flash
- **Do you see any meaningful error information in the DevTools?** 

Flash reported as failed despite success on two different Windows 11 machines

Error

gui.js:37 Error: Error invoking remote method 'enable-screensaver': No handler registered for 'enable-screensaver'
    at o.invoke (node:electron/js2c/renderer_init:57:526)

lurch, 2024-01-09:
See also https://github.com/balena-io/etcher/issues?q=is%3Aissue+is%3Aopen+screensaver :wink:

phillipremaker, 2024-01-09:
Thanks, @lurch! Should I mark as duplicate? Is there a master case for this? Is it recognized as an issue?

lurch, 2024-01-09:
@phillipremaker I don't know, I don't work on Etcher.

aethernet, 2024-01-23:
The `enable-screensaver` error is unrelated to what you describe.

Your problem might come from not being able to unmount the disks at the end of the process, hard to tell.

phillipremaker, 2024-01-25:
Thanks @aethernet! Any suggestions of how to do a diagnostic around that? I can't tell why the code has decided the flash process had failed.

aethernet, 2024-01-26:
I got the same issue yesterday on one of my test flash, but I don't have a solid reproduction yet

I'm suspecting it has to do with the OS mounting the drive before the checks are done, but idk yet how to confirm.

phillipremaker, 2024-02-02:
More details: Fails on the PCI flash on the motherboard, works on a USB-C reader. So the issue seems to be related to the flash hardware, possibly the unmount. The screensaver error is superfluous, as @aethernet noted. Will do some more tests.

staffa, 2024-04-30:
Just to add on and clarify, indeed, the failed message is not from the screensaver issue as aethernet suggests: you can note that there is an earlier error in the console pre-flash that does not get reported as failure, at least not at the time.  One should instead open, on the line just above, the Flash results object > results > errors > 0 to see the failure that the program is actually reporting.

In my case, at least, this error is a failure in the checksum. It's unclear to me if etcher has another method besides pure checksumming that would be useful to determine when in time or on disk the expected values diverge, but if it did and it would be helpful for those to be reported, please let us know. (Or alternatively if we should be directing to a different issue)

phillipremaker, 2024-04-30:
In a previous life, I had different flash hardware report a different checksum because one used an 8-bit read boundary and one a 16-bit read boundary, and the 16-bit hardware produced a different checksum on odd-byte sizes images since it included 8 bits of superfluous data not in the file on a checksum. I know nothing about how checksums are calculated in Etcher or if the OS abstraction hides that kind of thing, just a thought in case USB and PCIe hardware drivers do the magick differently.