balena-io/etcher #4164: Malwarebytes is flagging balenaetcher as malware

nascentt, 2024-01-21:
- **Etcher version:1.18.13.0** 
- **Operating system and architecture:Windows 10** 
- **Image flashed:** 
- **What do you think should have happened:** 
- **What happened:malwarebytes is flagging balena etcher as malware 'Malware.AI.3253061021'**
- **Do you see any meaningful error information in the DevTools?**

pelicanmedia, 2024-01-21:
Same here. Both quarantined.
[image]

aethernet, 2024-01-23:
Looks like false positive found by their AI system.

Can you check if it also happens with 1.19.3 (pre-release)?

pelicanmedia, 2024-01-24:
There is no virus warning after a new scan, but I get the below when installing 1.19.3...
[image]

aethernet, 2024-01-26:
I have been able to reproduced. This is actually a bug in the installer and not in etcher itself. If you let it go, it will eventually install and launch etcher.

I'm currently trying to find the cause.

Olman62, 2024-04-07:
The application made my 2 flash drives as write-protected and I cannot format them.

closed 2024-05-30 by aethernet