balena-io/etcher #4165: Error invoking remote method 'enable-screensaver'

CrazyWolf13, 2024-01-22:
- **Etcher version:** 
1.18.3
Edit: I tried the Pre-Release but they won't even install due to some nodejs missing, however nodejs and all in installed. so please don't suggest that.

- **Operating system and architecture:** 
Win1123H2
[image]

- **Image flashed:** 
 Proxmox ISO, but happens with ALL

- **What do you think should have happened:** 
No error, maybe a warning.

- **What happened:**
Flash seems to fail however I think the Error does not stop the usb from working.

- **Do you see any meaningful error information in the DevTools?** 
```
    at a.invoke (node:electron/js2c/renderer_init:2:7723)
(anonymous) @ gui.js:3
gui.js:1  0 devices, 0% at 0.00 MB/s (total 0.00 MB/s)  with 0 failed devices
gui.js:3 Successfully connected to IPC server: etcher-server-29496-1705947625490-privileged}}, socket root C:\Users\tobia\AppData\Local\Temp\
gui.js:3 params Object
gui.js:3 Image: C:\Users\tobia\Downloads\proxmox-ve_8.1-1.iso
gui.js:3 Devices: \\.\PhysicalDrive6
gui.js:3 Auto blockmapping: true
gui.js:3 Decompress first: true
gui.js:1 Flashing 1 device, 1% at 12.80 MB/s (total 12.80 MB/s) eta in 100s with 0 failed devices
--------------------------------------------------------------------------------------------------------------
gui.js:1 Verifying 1 device, 95% at 98.60 MB/s (total 98.60 MB/s) eta in 1s with 0 failed devices
gui.js:1 Verifying 1 device, 100% at 102.90 MB/s (total 102.90 MB/s) eta in 0s with 0 failed devices
gui.js:3 fail event
gui.js:3 Object
gui.js:3 Object
gui.js:3 Flash results Object
gui.js:3 Object
gui.js:3 Terminating IPC server
gui.js:3 Error: The writer process ended unexpectedly
    at createError (gui.js:3:225919)
    at Object.createUserError (gui.js:3:226132)
    at finish (gui.js:307:2613)
    at Server.onFail (gui.js:307:2100)
    at Server.emit (gui.js:3:445008)
    at Server.gotData (gui.js:3:450367)
    at Socket.emit (node:events:513:28)
    at addChunk (node:internal/streams/readable:324:12)
    at readableAddChunk (node:internal/streams/readable:293:11)
    at Readable.push (node:internal/streams/readable:234:10)
    at Pipe.onStreamRead (node:internal/stream_base_commons:190:23)
(anonymous) @ gui.js:3
gui.js:3 Error: Error invoking remote method 'enable-screensaver': Error: No handler registered for 'enable-screensaver'
    at a.invoke (node:electron/js2c/renderer_init:2:7723)
(anonymous) @ gui.js:3
```

lurch, 2024-01-23:
See also https://github.com/balena-io/etcher/issues?q=is%3Aissue+screensaver

CrazyWolf13, 2024-01-23:
Closing as duplicate in favour of 
#3957
#4161

closed 2024-01-23 by CrazyWolf13