balena-io/etcher #4170: error: (0 , hrequestMetadata) is not a function

0n1cOn3, 2024-01-29:
- **Etcher version:** 
1.19.1
- **Operating system and architecture:** 
Fedora 39, x86-x64
- **Image flashed:**
FreeDOS 
- **What do you think should have happened:** 
Passwort request for root access to the hardware to beginn with the flashing process
- **What happened:**
error: (0 , hrequestMetadata) is not a function Balena Etcher Error Screen appears
- **Do you see any meaningful error information in the DevTools?** 

[222333:0129/140123.475498:ERROR:bus.cc(407)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[222333:0129/140123.475614:ERROR:bus.cc(407)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[222333:0129/140123.475638:ERROR:bus.cc(407)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[222333:0129/140123.475658:ERROR:bus.cc(407)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[222333:0129/140123.493905:ERROR:bus.cc(407)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
Language changed to: en
[222333:0129/140123.992979:ERROR:bus.cc(407)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
ready-to-show: 567.283ms
packageUpdatable false
[222364:0129/140125.601180:ERROR:gl_surface_presentation_helper.cc(260)] GetVSyncParametersIfAvailable() failed for 1 times!
[222364:0129/140127.475290:ERROR:gl_surface_presentation_helper.cc(260)] GetVSyncParametersIfAvailable() failed for 2 times!
[222364:0129/140127.486597:ERROR:gl_surface_presentation_helper.cc(260)] GetVSyncParametersIfAvailable() failed for 3 times!
Warning: /usr/lib/libvulkan_intel.so: falsche ELF-Klasse: ELFCLASS32
Warning: loader_icd_scan: Failed loading library associated with ICD JSON /usr/lib/libvulkan_intel.so. Ignoring this JSON
Warning: /usr/lib/libvulkan_radeon.so: falsche ELF-Klasse: ELFCLASS32
Warning: loader_icd_scan: Failed loading library associated with ICD JSON /usr/lib/libvulkan_radeon.so. Ignoring this JSON
Warning: /usr/lib/libvulkan_broadcom.so: falsche ELF-Klasse: ELFCLASS32
Warning: loader_icd_scan: Failed loading library associated with ICD JSON /usr/lib/libvulkan_broadcom.so. Ignoring this JSON
Warning: /usr/lib/libvulkan_panfrost.so: falsche ELF-Klasse: ELFCLASS32
Warning: loader_icd_scan: Failed loading library associated with ICD JSON /usr/lib/libvulkan_panfrost.so. Ignoring this JSON
Warning: /usr/lib/libvulkan_lvp.so: falsche ELF-Klasse: ELFCLASS32
Warning: loader_icd_scan: Failed loading library associated with ICD JSON /usr/lib/libvulkan_lvp.so. Ignoring this JSON
Warning: /usr/lib/libvulkan_intel_hasvk.so: falsche ELF-Klasse: ELFCLASS32
Warning: loader_icd_scan: Failed loading library associated with ICD JSON /usr/lib/libvulkan_intel_hasvk.so. Ignoring this JSON
Warning: /usr/lib/libvulkan_freedreno.so: falsche ELF-Klasse: ELFCLASS32
Warning: loader_icd_scan: Failed loading library associated with ICD JSON /usr/lib/libvulkan_freedreno.so. Ignoring this JSON
error: XDG_RUNTIME_DIR is invalid or not set in the environment.
error: XDG_RUNTIME_DIR is invalid or not set in the environment.
[222333:0129/140447.417550:ERROR:CONSOLE(38)] "console.assert", source: devtools://devtools/bundled/panels/application/application.js (38)

lurch, 2024-01-29:
Duplicate of #4150 ?

0n1cOn3, 2024-01-29:
> Duplicate of #4150 ?

Yeah, odd that Github hasn't show me that when I have looked up the same topic name, that this already has been opened by someone else 🥲

lurch, 2024-01-29:
> odd that Github hasn't show me that when I have looked up the same topic name

That other issue has a `.` between `h` and `requestMetadata` :wink:

0n1cOn3, 2024-01-29:
> > odd that Github hasn't show me that when I have looked up the same topic name
> 
> 
> 
> That other issue has a `.` between `h` and `requestMetadata` :wink: 

Ahh!! Alright 😇

closed 2024-01-29 by 0n1cOn3