balena-io/etcher #4171: Downloading the macOS version from your website gives me an appimage file that can not be opened.

b-hayes, 2024-01-31:
- **Etcher version:** 
- latest
- **Operating system and architecture:** 
- macos sonoma
- **Image flashed:** 
- na
- **What do you think should have happened:** 
- I should probably receive a file that works on macos given that is the version of the app I am atetmpting to download form you.
- **What happened:**
- I get some trange file that cant be used on macos
- **Do you see any meaningful error information in the DevTools?** 
- no

dfunckt, 2024-01-31:
Where are you trying to download etcher from? Can you share a link to that website?

As it is, our website shows a list of options and the user gets to pick what to download based on their platform and preference. Which leads me to believe you either try to download etcher from one of the many fake sites. or just select the wrong platform option.

lurch, 2024-01-31:
(AFAIK, AppImage files are only used on Linux and not macOS or Windows?)

b-hayes, 2024-03-01:
It was coming from the official websites macos x64 link. Seems I get a dmg file when i try it now tho so 🤷🏼‍♂️

closed 2024-03-01 by dfunckt

mullapuh, 2024-03-20:
Dont think this is solved. Just tried to download and result is the same

lurch, 2024-03-20:
Do the downloads [here](https://github.com/balena-io/etcher/releases) work properly?

wlegate, 2024-03-30:
It's not working for me either on MacOS 14.4.X @lurch… it downloads & seemingly installs fine, but no interface actually appears when you open the app. I downloaded it two separate times from the official MacOS install link.

b-hayes, 2024-04-03:
@wlegate sounds like a different problem. I wasn't getting an installable file it was for the wrong platform entirely. You should probably open a new issue.