balena-io/etcher #4174: Uploaded IMG file = next steps greyed-out

m8tterfact, 2024-02-09:
- **Etcher version:** 
- **Operating system and architecture:** 
- **Image flashed:** 
- **What do you think should have happened:** 
- **What happened:**
- **Do you see any meaningful error information in the DevTools?**

m8tterfact, 2024-02-09:
Once OI either drag and drop or select .img file to be made bootable, the next options selecting drive etc are greyed-out so you cannot select or advance - I tried to give it some time 10 min... to no avaiul

Alex313031, 2024-03-16:
@m8tterfact It's this same bug > https://github.com/balena-io/etcher/issues/4150

Eric-FR, 2024-07-15:
I'm not sure that it is the same bug. Using 1.19.21, I first faced the other bug. Then, I went to 1.8.13. So, when selecting the img file, nothing happen. The select file button is greyed. The select support also. I waited for more than one hour. No change. Indeed no activity on monitoring.

Windows 10 familly. 22H2.

lurch, 2024-07-16:
> The select file button is greyed. The select support also.

Sounds like the same problem as #4267 ?

Eric-FR, 2024-07-16:
Yes, it looks like being the same problem.

Eric-FR, 2024-07-16:
Back to 1.18.11, I was able to read the file and flash it.