balena-io/etcher #4181: PPA or Deb Repo?

InklingGirl, 2024-02-22:
I would like it if I could auto-update this app, but it's not packed in my distro's (Ubuntu's) repos, the Snap Store, or Flathub. However, I've seen references online – [including in the README](https://github.com/balena-io/etcher?tab=readme-ov-file#debian-and-ubuntu-based-package-repository-gnulinux-x86x64) – to a repo for it, w/ some of these explicitly mentioning it's for deployment/updating. Is there such a repo? How can I index it to start updating from it? Thank you.

lurch, 2024-02-22:
See the sticky issue: #4059

InklingGirl, 2024-02-23:
well, it's certainly been more than a few weeks since that announcement.

InklingGirl, 2024-05-11:
Revisiting this, still not at all sure how to add a repo for auto-updating of a native .deb of `balena-etcher`.

lurch, 2024-05-12:
> how the heck do I make an `apt` entry for the repo?

Currently, I don't think you can. You need to download and install the `.deb` file manually.