balena-io/etcher #4186: Fedora [1.19.5-1.x86_64] "Error Opening Source: (0, h.requestMetadata) is not a function"

eengmann, 2024-03-15:
- **Etcher version:** balena-etcher-1.19.5-1.x86_64
- **Operating system and architecture:** fedora fc38.x86_64
- **Image flashed:** CentOS-7-aarch64-Minimal-2009.iso
- **What do you think should have happened:**
- After clicking "Flash from file" from the main GUI, and the selecting the iso image, I expected the program to continue with the guided USB-flashing process. Using the iso download link with "Flash from URL" instead, I ran into the same error message.
- **What happened:**
- Selecting the iso file and clicking on "Open" only resulted in the aforementioned error message box. After acknowledging the error, I returned to the main menu.
- **Do you see any meaningful error information in the DevTools?** 

index.js:2 TypeError: (0 , h.requestMetadata) is not a function
    at index.js:140:1418
    at H.selectSource (index.js:140:2212)
    at H.openImageSelector (index.js:140:2543)
(anonymous) @ index.js:2

Alex313031, 2024-03-16:
@eengmann Same here, and also with building on Windows > https://github.com/balena-io/etcher/issues/4150

eengmann, 2024-03-16:
Same as issue  #4150

eengmann, 2024-03-16:
Same as issue #4150

closed 2024-03-16 by eengmann