balena-io/etcher #4200: Won't build Apple Silicon binary, no existing Apple Silicon binary in the Releases

matthewyang204, 2024-04-05:
- **Operating system and architecture: Apple Silicon (Apple's custom ARM) Mac
- **What happened: Fails to build

I tried to build a native Apple Silicon binary from your source code using NPM. I get a bunch of errors, listed below.
npm install:
```
npm WARN EBADENGINE Unsupported engine {
npm WARN EBADENGINE   package: 'balena-etcher@1.19.5',
npm WARN EBADENGINE   required: { node: '>=18 <20' },
npm WARN EBADENGINE   current: { node: 'v21.7.2', npm: '10.5.0' }
npm WARN EBADENGINE }
npm WARN EBADENGINE Unsupported engine {
npm WARN EBADENGINE   package: 'drivelist@11.1.0',
npm WARN EBADENGINE   required: { node: '>=16 < 19' },
npm WARN EBADENGINE   current: { node: 'v21.7.2', npm: '10.5.0' }
npm WARN EBADENGINE }
npm WARN skipping integrity check for git dependency ssh://git@github.com/balena-io-modules/unbzip2-stream.git 
npm WARN cleanup Failed to remove some directories [
npm WARN cleanup   [
npm WARN cleanup     '/Users/matthewyang/Documents/Code/Xcode/etcher/node_modules/@ronomon/direct-io',
npm WARN cleanup     [Error: ENOTEMPTY: directory not empty, rmdir '/Users/matthewyang/Documents/Code/Xcode/etcher/node_modules/@ronomon/direct-io'] {
npm WARN cleanup       errno: -66,
npm WARN cleanup       code: 'ENOTEMPTY',
npm WARN cleanup       syscall: 'rmdir',
npm WARN cleanup       path: '/Users/matthewyang/Documents/Code/Xcode/etcher/node_modules/@ronomon/direct-io'
npm WARN cleanup     }
npm WARN cleanup   ],
npm WARN cleanup   [
npm WARN cleanup     '/Users/matthewyang/Documents/Code/Xcode/etcher/node_modules/@ronomon/direct-io',
npm WARN cleanup     [Error: ENOTEMPTY: directory not empty, rmdir '/Users/matthewyang/Documents/Code/Xcode/etcher/node_modules/@ronomon/direct-io'] {
npm WARN cleanup       errno: -66,
npm WARN cleanup       code: 'ENOTEMPTY',
npm WARN cleanup       syscall: 'rmdir',
npm WARN cleanup       path: '/Users/matthewyang/Documents/Code/Xcode/etcher/node_modules/@ronomon/direct-io'
npm WARN cleanup     }
npm WARN cleanup   ]
npm WARN cleanup ]
npm ERR! code 1
npm ERR! path /Users/matthewyang/Documents/Code/Xcode/etcher/node_modules/mountutils
npm ERR! command failed
npm ERR! command sh -c prebuild-install || node-gyp rebuild
npm ERR! CXX(target) Release/obj.target/MountUtils/src/mountutils.o
npm ERR! gyp info it worked if it ends with ok
npm ERR! gyp info using node-gyp@9.4.0
npm ERR! gyp info using node@21.7.2 | darwin | arm64
npm ERR! gyp info find Python using Python version 3.12.2 found at "/opt/homebrew/opt/python@3.12/bin/python3.12"
npm ERR! gyp http GET https://nodejs.org/download/release/v21.7.2/node-v21.7.2-headers.tar.gz
npm ERR! gyp http 200 https://nodejs.org/download/release/v21.7.2/node-v21.7.2-headers.tar.gz
npm ERR! gyp http GET https://nodejs.org/download/release/v21.7.2/SHASUMS256.txt
npm ERR! gyp http 200 https://nodejs.org/download/release/v21.7.2/SHASUMS256.txt
npm ERR! gyp info spawn /opt/homebrew/opt/python@3.12/bin/python3.12
npm ERR! gyp info spawn args [
npm ERR! gyp info spawn args   '/Users/matthewyang/Documents/Code/Xcode/etcher/node_modules/node-gyp/gyp/gyp_main.py',
npm ERR! gyp info spawn args   'binding.gyp',
npm ERR! gyp info spawn args   '-f',
npm ERR! gyp info spawn args   'make',
npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   '/Users/matthewyang/Documents/Code/Xcode/etcher/node_modules/mountutils/build/config.gypi',
npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   '/Users/matthewyang/Documents/Code/Xcode/etcher/node_modules/node-gyp/addon.gypi',
npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   '/Users/matthewyang/Library/Caches/node-gyp/21.7.2/include/node/common.gypi',
npm ERR! gyp info spawn args   '-Dlibrary=shared_library',
npm ERR! gyp info spawn args   '-Dvisibility=default',
npm ERR! gyp info spawn args   '-Dnode_root_dir=/Users/matthewyang/Library/Caches/node-gyp/21.7.2',
npm ERR! gyp info spawn args   '-Dnode_gyp_dir=/Users/matthewyang/Documents/Code/Xcode/etcher/node_modules/node-gyp',
npm ERR! gyp info spawn args   '-Dnode_lib_file=/Users/matthewyang/Library/Caches/node-gyp/21.7.2/<(target_arch)/node.lib',
npm ERR! gyp info spawn args   '-Dmodule_root_dir=/Users/matthewyang/Documents/Code/Xcode/etcher/node_modules/mountutils',
npm ERR! gyp info spawn args   '-Dnode_engine=v8',
npm ERR! gyp info spawn args   '--depth=.',
[... 111 lines cut ...]
npm ERR! gyp ERR! build error 
npm ERR! gyp ERR! stack Error: `make` failed with exit code: 2
npm ERR! gyp ERR! stack     at ChildProcess.onExit (/Users/matthewyang/Documents/Code/Xcode/etcher/node_modules/node-gyp/lib/build.js:203:23)
npm ERR! gyp ERR! stack     at ChildProcess.emit (node:events:519:28)
npm ERR! gyp ERR! stack     at ChildProcess._handle.onexit (node:internal/child_process:294:12)
npm ERR! gyp ERR! System Darwin 23.4.0
npm ERR! gyp ERR! command "/opt/homebrew/Cellar/node/21.7.2/bin/node" "/Users/matthewyang/Documents/Code/Xcode/etcher/node_modules/.bin/node-gyp" "rebuild"
npm ERR! gyp ERR! cwd /Users/matthewyang/Documents/Code/Xcode/etcher/node_modules/mountutils
npm ERR! gyp ERR! node -v v21.7.2
npm ERR! gyp ERR! node-gyp -v v9.4.0
npm ERR! gyp ERR! not ok

npm ERR! A complete log of this run can be found in: /Users/matthewyang/.npm/_logs/2024-04-05T19_22_53_424Z-debug-0.log
```

When I try to run npm run build --arch=arm64:
```
npm ERR! Missing script: "build"
npm ERR! 
npm ERR! To see a list of scripts, run:
npm ERR!   npm run

npm ERR! A complete log of this run can be found in: /Users/matthewyang/.npm/_logs/2024-04-05T19_21_20_166Z-debug-0.log
```
Also, I'm kind of annoyed that there isn't a native Apple Silicon binary already in the Releases. Like what's so hard about it?

KwisatzJim, 2024-04-23:
I do see one issue that may be part of the problem. At the top of the output there's this:

```
npm WARN EBADENGINE Unsupported engine {
npm WARN EBADENGINE   package: 'balena-etcher@1.19.5',
npm WARN EBADENGINE   required: { node: '>=18 <20' },
npm WARN EBADENGINE   current: { node: 'v21.7.2', npm: '10.5.0' }
npm WARN EBADENGINE }
npm WARN EBADENGINE Unsupported engine {
npm WARN EBADENGINE   package: 'drivelist@11.1.0',
npm WARN EBADENGINE   required: { node: '>=16 < 19' },
npm WARN EBADENGINE   current: { node: 'v21.7.2', npm: '10.5.0' }
npm WARN EBADENGINE }

```
so it looks like you've got node v21.7.2 which is higher version than supported by etcher. If you installed node through brew, with brew install node, that installs v21.7.2.  You can try uninstalling that then install a version within the limits listed, which is at least v18 but less than v20... 

`brew install node@18
`

I've had that work, but with the latest git pull been getting other errors

aethernet, 2024-04-25:
current master builds on node 20

aethernet, 2024-04-25:
You can also try the latest pre-release which have an apple silicon build

luckman212, 2024-05-28:
It's just a few weeks shy of a year since a non-beta release has posted. How are these [1.19.x betas](https://github.com/balena-io/etcher/releases) running? Any reason NOT to use them vs. the last "official" release [v1.18.11](https://github.com/balena-io/etcher/releases/tag/v1.18.11) ?

matthewyang204, 2024-05-29:
I'll look into the beta releases. I have run into several problems before
with beta releases, I'll try to see if I like these new beta releases.

matthewyang204, 2024-06-05:
A new Apple Silicon binary was released, closing this thread.

closed 2024-06-05 by matthewyang204