balena-io/etcher #4201: "A JavaScript Error Occured in the Main Process" Error on startup

SemihGuner, 2024-04-10:
- **Etcher version:** 1.19.5
- **Operating system and architecture:** Windows 10 x64
- **Image flashed:** Nobara Official.
- **What do you think should have happened:**  I just opened the app, and the error popped out. 
- **What happened:** The error says that there's "A JavaScript Error Occured in the Main Process". When I clicked OK, the app continued to run normally. I don't know if it broke anything inside the app, though. 
- **Do you see any meaningful error information in the DevTools?**  No, I currently don't.
[image]

Alex313031, 2024-04-12:
@SemihGuner Yup. They need to fix it. I have fixed it in my fork, by simply editing one line. They didn't do import correctly.

aethernet, 2024-04-25:
@Alex313031 what was the fix?

aethernet, 2024-04-25:
nvm found it :)
https://github.com/balena-io/etcher/pull/4209

closed 2024-04-25 by aethernet