balena-io/etcher #4206: fails building on Apple Silicon with error "You must depend on "electron-prebuilt-compile" in your devDependencies"

KwisatzJim, 2024-04-24:
- **Etcher version:** 
- 1.19.10
- **Operating system and architecture:** 
- macOS 14.4.1
- Macbook Pro M3 Pro
- **Image flashed:** 
- n/a
- **What do you think should have happened:** 
- brew install node@20
- sudo npm install node-gyp
- sudo npm install electron
- sudo npm install electron-forge
- sudo npm install jq
- sudo npm run-script make
Password:

> balena-etcher@1.19.10 make
> electron-forge make

✔ Checking your system
✖ Resolving Forge Config

Electron forge was terminated:
You must depend on "electron-prebuilt-compile" in your devDependencies
- **What happened:**
- fails the build process
- **Do you see any meaningful error information in the DevTools?**

matthewyang204, 2024-04-24:
I got a similar error, except it was "electron-builder". I've been trying to build balenaEtcher on Apple Silicon for 2 years now. My Mac keeps on getting different errors, and the latest error I've got is that. I don't know what it is about balenaEtcher, being extremely hard to build balenaEtcher on Apple Silicon.

aethernet, 2024-04-25:
I have no clue what's happening.

I'm building multiple times every days on a m1 without any issue and there's nothing special about my setup
$ npm i 
$ npm make

KwisatzJim, 2024-04-25:
Interesting, running npm run make works but npm run-scripts make doesn't, while it previously did. Perhaps i missed an update/change along the way. 

Thanks!

closed 2024-04-25 by KwisatzJim