balena-io/etcher #4215: Mac crashes during validation

LiminalD, 2024-04-27:
Hi,

My Mac crashes during validation after flashing to the SD Card. I haven’t encountered this issue before; it just started happening last week. I’m running macOS 14.1.1 Apple m1 and using BalenaEtcher 1.19.16

I’ve tried granting full disk permission and moving the image to the .app directory, but nothing seems to be working. Any ideas on how to solve this issue?

panic(cpu 0 caller 0xfffffe0019488cf8): "dart-apcie1 (0xfffffe20064a0000): DART(DART) error: SID 1 PTE invalid exception on write of DVA 0xad8000 (TTBR 0 SEG 0 PTE 0x2b6) ERROR_STATUS 0x81000404 TIME 0x199203cca44 TTE 0 AXI_ID 0" @AppleT6000DART.cpp:1407
Debugger message: panic
Memory ID: 0x6
OS release type: User
OS version: 23E224
Kernel version: Darwin Kernel Version 23.4.0: Fri Mar 15 00:10:42 PDT 2024; root:xnu-10063.101.17~1/RELEASE_ARM64_T6000
Fileset Kernelcache UUID: 9EC7FBEFD78CA9C9BCFBF3637BA3DB70
Kernel UUID: 52F71D8C-FEDF-3A4B-9B6B-8463C5251B20
Boot session UUID: 447C73BB-4DE1-4EF1-9F38-A1AB31057756
iBoot version: iBoot-10151.101.3
secure boot?: YES
roots installed: 0
Paniclog version: 14
KernelCache slide: 0x000000000f638000
KernelCache base:  0xfffffe001663c000
Kernel slide:      0x000000000f640000
Kernel text base:  0xfffffe0016644000
Kernel text exec slide: 0x0000000010b70000
Kernel text exec base:  0xfffffe0017b74000
mach_absolute_time: 0x9d1b8f99c
Epoch Time:        sec       usec
  Boot    : 0x662cdc53 0x000cbf3c
  Sleep   : 0x00000000 0x00000000
  Wake    : 0x00000000 0x00000000
  Calendar: 0x662ce32a 0x0003514d

Zone info:
  Zone map: 0xfffffe100626c000 - 0xfffffe300626c000
  . VM    : 0xfffffe100626c000 - 0xfffffe14d2f38000
  . RO    : 0xfffffe14d2f38000 - 0xfffffe166c8d0000
  . GEN0  : 0xfffffe166c8d0000 - 0xfffffe1b3959c000
  . GEN1  : 0xfffffe1b3959c000 - 0xfffffe2006268000
  . GEN2  : 0xfffffe2006268000 - 0xfffffe24d2f34000
  . GEN3  : 0xfffffe24d2f34000 - 0xfffffe299fc00000
  . DATA  : 0xfffffe299fc00000 - 0xfffffe300626c000
  Metadata: 0xfffffe3f426cc000 - 0xfffffe3f4a6cc000
  Bitmaps : 0xfffffe3f4a6cc000 - 0xfffffe3f4d3f8000
  Extra   : 0 - 0

TPIDRx_ELy = {1: 0xfffffe1b387fc800  0: 0x0000000000000000  0ro: 0x000000016de670e0 }
CORE 0 PVH locks held: None
CORE 1 PVH locks held: None
CORE 2 PVH locks held: None
CORE 3 PVH locks held: None
CORE 4 PVH locks held: None
CORE 5 PVH locks held: None
CORE 6 PVH locks held: None
CORE 7 PVH locks held: None
CORE 8 PVH locks held: None
CORE 9 PVH locks held: None
CORE 0 is the one that panicked. Check the full backtrace for details.
CORE 1: PC=0xfffffe0018348ebc, LR=0xfffffe001a0c6710, FP=0xfffffe8e130e6a00
CORE 2: PC=0xfffffe0017c098c0, LR=0xfffffe0017c098bc, FP=0xfffffe8e12e6bef0
CORE 3: PC=0xfffffe0017c098c0, LR=0xfffffe0017c098bc, FP=0xfffffe8e12927ef0
CORE 4: PC=0x00000001098ec5f4, LR=0x00000001098ec300, FP=0x000000020881e320
CORE 5: PC=0xfffffe0017c098c0, LR=0xfffffe0017c098bc, FP=0xfffffe8e129cfef0
CORE 6: PC=0xfffffe0017c098c0, LR=0xfffffe0017c098bc, FP=0xfffffe8e12a6bef0
CORE 7: PC=0xfffffe0017c098c0, LR=0xfffffe0017c098bc, FP=0xfffffe8e12f17ef0
CORE 8: PC=0xfffffe0017c098c0, LR=0xfffffe0017c098bc, FP=0xfffffe8e11967ef0
CORE 9: PC=0xfffffe0017c098c0, LR=0xfffffe0017c098bc, FP=0xfffffe8e12a17ef0
Compressor Info: 3% of compressed pages limit (OK) and 4% of segments limit (OK) with 1 swapfiles and OK swap space
Panicked task 0xfffffe1b39581018: 6976 pages, 19 threads: pid 402: WindowServer
Panicked thread: 0xfffffe1b387fc800, backtrace: 0xfffffe6fc196b5a0, tid: 4397
		  lr: 0xfffffe0017bcdf70  fp: 0xfffffe6fc196b630
[... 251 lines cut ...]
com.apple.iokit.IOAVFamily	1.0.0
com.apple.iokit.IOHDCPFamily	1.0.0
com.apple.iokit.IOCECFamily	1
com.apple.iokit.IOAudio2Family	1.0
com.apple.driver.AppleIISController	340.1
com.apple.driver.AppleAudioClockLibs	340.8
com.apple.driver.FairPlayIOKit	71.10.0
com.apple.driver.AppleARMPlatform	1.0.2
com.apple.iokit.IOSlowAdaptiveClockingFamily	1.0.0
com.apple.iokit.IOReportFamily	47
com.apple.security.quarantine	4
com.apple.security.sandbox	300.0
com.apple.iokit.IOStorageFamily	2.1
com.apple.kext.AppleMatch	1.0.0d1
com.apple.driver.AppleMobileFileIntegrity	1.0.5
com.apple.iokit.CoreAnalyticsFamily	1
com.apple.security.AppleImage4	6.3.0
com.apple.kext.CoreTrust	1
com.apple.iokit.IOCryptoAcceleratorFamily	1.0.1
com.apple.kec.pthread	1
com.apple.kec.Libm	1
com.apple.kec.Compression	1.0
com.apple.kec.corecrypto	14.0

** Stackshot Succeeded ** Bytes Traced 549105 (Uncompressed 1448480) **

mochsner, 2024-05-13:
@LiminalD possibly unrelated, but incase there's a mac driver issue on Sonoma 14.4.1, I'll note that my wife's (bought used) macbook pro m2 is crashing with this same message when she copies files from the SD card (photos) onto her desktop. 

Do you happen to get any 'permissions' error type popups sometimes completing this workflow too? Guessing not, but worth checking

LiminalD, 2024-05-13:
Hi,

After investigation and a complete reinstall of my Mac, it continues to crash. It appears that many card readers in MacBook Pro M1 (and apparently also M2) models are defective. This is a known issue at Apple; however, they have been very silent about it. The temporary solution is to use an external card reader, which works flawlessly.

denisbondar, 2024-05-31:
Hi,

I seem to have the same problem.
I am using on MacBook M1 Pro, OS 14.5 Sonoma and Promate USB-HUB.

While writing to USB-Stick, a write error occurs at random places. Sometimes the write error occurs at the validation stage, either immediately at the start of validation or shortly after the start.
However, recording to SD card and verification using this USB-HUB is successful.

Can I provide some additional information to find the problem?