balena-io/etcher #4217: Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")

jiapei100, 2024-04-29:
- **Etcher version:**  v1.19.16
- **Operating system and architecture:**  Ubuntu 24.04
- **Image flashed:**  Nope 
- **What do you think should have happened:**  **AppImage** should be able to run.
- **What happened:** errors as follows
- **Do you see any meaningful error information in the DevTools?** 

```console
➜  Tools sudo ./balenaEtcher-1.19.16-x64.AppImage --no-sandbox
[sudo] password for lvision: 
[9921:0429/041126.485222:ERROR:bus.cc(407)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[9921:0429/041126.485302:ERROR:bus.cc(407)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[9921:0429/041126.485354:ERROR:bus.cc(407)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[9921:0429/041126.485373:ERROR:bus.cc(407)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
Authorization required, but no authorization protocol specified

Authorization required, but no authorization protocol specified

Authorization required, but no authorization protocol specified

Authorization required, but no authorization protocol specified

[9921:0429/041126.757713:ERROR:bus.cc(407)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
Authorization required, but no authorization protocol specified

Authorization required, but no authorization protocol specified

[9957:0429/041126.852550:ERROR:angle_platform_impl.cc(44)] Display.cpp:1086 (initialize): ANGLE Display::initialize error 12289: Could not open the default X display.
ERR: Display.cpp:1086 (initialize): ANGLE Display::initialize error 12289: Could not open the default X display.
[9957:0429/041126.854991:ERROR:gl_display.cc(515)] EGL Driver message (Critical) eglInitialize: Could not open the default X display.
[9957:0429/041126.855041:ERROR:gl_display.cc(786)] eglInitialize OpenGL failed with error EGL_NOT_INITIALIZED, trying next display type
Authorization required, but no authorization protocol specified

[9957:0429/041126.855249:ERROR:angle_platform_impl.cc(44)] Display.cpp:1086 (initialize): ANGLE Display::initialize error 12289: Could not open the default X display.
ERR: Display.cpp:1086 (initialize): ANGLE Display::initialize error 12289: Could not open the default X display.
[9957:0429/041126.855302:ERROR:gl_display.cc(515)] EGL Driver message (Critical) eglInitialize: Could not open the default X display.
[9957:0429/041126.855342:ERROR:gl_display.cc(786)] eglInitialize OpenGLES failed with error EGL_NOT_INITIALIZED
[9957:0429/041126.855381:ERROR:gl_display.cc(820)] Initialization of all EGL display types failed.
[9957:0429/041126.855423:ERROR:gl_ozone_egl.cc(26)] GLDisplayEGL::Initialize failed.
Authorization required, but no authorization protocol specified

Authorization required, but no authorization protocol specified

[9957:0429/041127.018818:ERROR:angle_platform_impl.cc(44)] Display.cpp:1086 (initialize): ANGLE Display::initialize error 12289: Could not open the default X display.
ERR: Display.cpp:1086 (initialize): ANGLE Display::initialize error 12289: Could not open the default X display.
[9957:0429/041127.018957:ERROR:gl_display.cc(515)] EGL Driver message (Critical) eglInitialize: Could not open the default X display.
[9957:0429/041127.019014:ERROR:gl_display.cc(786)] eglInitialize OpenGL failed with error EGL_NOT_INITIALIZED, trying next display type
Authorization required, but no authorization protocol specified

[9957:0429/041127.019246:ERROR:angle_platform_impl.cc(44)] Display.cpp:1086 (initialize): ANGLE Display::initialize error 12289: Could not open the default X display.
ERR: Display.cpp:1086 (initialize): ANGLE Display::initialize error 12289: Could not open the default X display.
[9957:0429/041127.019310:ERROR:gl_display.cc(515)] EGL Driver message (Critical) eglInitialize: Could not open the default X display.
[9957:0429/041127.019361:ERROR:gl_display.cc(786)] eglInitialize OpenGLES failed with error EGL_NOT_INITIALIZED
[9957:0429/041127.019408:ERROR:gl_display.cc(820)] Initialization of all EGL display types failed.
[9957:0429/041127.019460:ERROR:gl_ozone_egl.cc(26)] GLDisplayEGL::Initialize failed.
[9957:0429/041127.021027:ERROR:viz_main_impl.cc(198)] Exiting GPU process due to errors during initialization
Authorization required, but no authorization protocol specified

Authorization required, but no authorization protocol specified

Authorization required, but no authorization protocol specified

Authorization required, but no authorization protocol specified

Authorization required, but no authorization protocol specified

Authorization required, but no authorization protocol specified

[9998:0429/041127.228813:ERROR:angle_platform_impl.cc(44)] DisplayVkXcb.cpp:58 (initialize): xcb_connect() failed, error 1
ERR: DisplayVkXcb.cpp:58 (initialize): xcb_connect() failed, error 1
[9998:0429/041127.228913:ERROR:angle_platform_impl.cc(44)] Display.cpp:1086 (initialize): ANGLE Display::initialize error 12289: Not initialized.
ERR: Display.cpp:1086 (initialize): ANGLE Display::initialize error 12289: Not initialized.
[9998:0429/041127.228954:ERROR:gl_display.cc(515)] EGL Driver message (Critical) eglInitialize: Not initialized.
[9998:0429/041127.228990:ERROR:gl_display.cc(786)] eglInitialize SwANGLE failed with error EGL_NOT_INITIALIZED
[9998:0429/041127.229030:ERROR:gl_display.cc(820)] Initialization of all EGL display types failed.
[9998:0429/041127.229071:ERROR:gl_ozone_egl.cc(26)] GLDisplayEGL::Initialize failed.
Authorization required, but no authorization protocol specified

Authorization required, but no authorization protocol specified

[9998:0429/041127.380647:ERROR:angle_platform_impl.cc(44)] DisplayVkXcb.cpp:58 (initialize): xcb_connect() failed, error 1
ERR: DisplayVkXcb.cpp:58 (initialize): xcb_connect() failed, error 1
[9998:0429/041127.380777:ERROR:angle_platform_impl.cc(44)] Display.cpp:1086 (initialize): ANGLE Display::initialize error 12289: Not initialized.
ERR: Display.cpp:1086 (initialize): ANGLE Display::initialize error 12289: Not initialized.
[9998:0429/041127.380849:ERROR:gl_display.cc(515)] EGL Driver message (Critical) eglInitialize: Not initialized.
[9998:0429/041127.380910:ERROR:gl_display.cc(786)] eglInitialize SwANGLE failed with error EGL_NOT_INITIALIZED
[9998:0429/041127.380973:ERROR:gl_display.cc(820)] Initialization of all EGL display types failed.
[9998:0429/041127.381031:ERROR:gl_ozone_egl.cc(26)] GLDisplayEGL::Initialize failed.
[9998:0429/041127.382494:ERROR:viz_main_impl.cc(198)] Exiting GPU process due to errors during initialization
Authorization required, but no authorization protocol specified

Authorization required, but no authorization protocol specified

[10007:0429/041127.421196:ERROR:x11_software_bitmap_presenter.cc(144)] XGetWindowAttributes failed for window 60817412
Language changed to: en
[9921:0429/041127.920026:ERROR:bus.cc(407)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
ready-to-show: 1.604s
packageUpdatable false

```

jiapei100, 2024-04-29:
What's more, I was trying to install the **.deb**, but got the following **ERROR** messages:

```console
➜  Tools sudo dpkg -i balena-etcher_1.19.16_amd64.deb
Selecting previously unselected package balena-etcher.
(Reading database ... 500082 files and directories currently installed.)
Preparing to unpack balena-etcher_1.19.16_amd64.deb ...
Unpacking balena-etcher (1.19.16) ...
dpkg: dependency problems prevent configuration of balena-etcher:
 balena-etcher depends on gconf-service; however:
  Package gconf-service is not installed.
 balena-etcher depends on gconf2; however:
  Package gconf2 is not installed.
 balena-etcher depends on libgconf-2-4; however:
  Package libgconf-2-4 is not installed.

dpkg: error processing package balena-etcher (--install):
 dependency problems - leaving unconfigured
Processing triggers for gnome-menus (3.36.0-1.1ubuntu3) ...
Processing triggers for desktop-file-utils (0.27-2build1) ...
Errors were encountered while processing:
 balena-etcher
```

Can anybody help? Thanks.

aethernet, 2024-05-01:
Hello, 

Thanks for the feedback.
Did you check the [required dependencies](https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#runtime-gnulinux-dependencies)? 

According to the error you got, you should be able to install the deb by first installing :`gconf-service gconf2 libgrconf-2-4`

You should also check [the Wayland entry](https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#running-on-wayland)