balena-io/etcher #4219: Can not compile etcher on Windows

Jeffson-wu, 2024-05-08:
Hi,
I need help to compile etcher by myself, can I ask you:
1:can you guide me how to compile it on windows?I did not find a detailed step in CONTRIBUTING.md.

2:it is possible to compile windows exe at ubuntu linux env?I tried to compile both windows and linux version of etcher, but both failed.

Thanks

aethernet, 2024-05-08:
Hello, 

Documentation might be a little behind as we changed quite some things recently.
You can check the https://github.com/balena-io/etcher/blob/master/.github/actions/test/action.yml for the most up-to-date process (that's the code that does the build for the public version).

We moved to `electron-forge` so as long as you comply with the prerequisite for electron and electron-forge (and installed the host packages listed in the package.json for ubuntu), you should be good to go with an `npm run package`.

Jeffson-wu, 2024-05-08:
Thank you @aethernet 
I checked the action.yml file, it sees like the dependencies for Windows is not mentioned.
If I want to build etcher for windows, I can build the code on linux and then package the executable file for running on windows, it this correct?

aethernet, 2024-05-08:
You can refer to the electron-forge docs for both answers.

I guess it's possible to compile for windows on Linux but didn't tried and don't know what dependencies you're going to need for that...

Jeffson-wu, 2024-05-12:
@aethernet 
Thanks for your kind help, the linux package build succeed after I installed electron-forge.
but for windows build, I am facing an error, if you guys know how to solve this error, it will be very helpful,but never mind if you guys do not know since I am using Ubuntu to build Windows package.

wu@wu-VirtualBox:~/etcher$ npm run make -- --platform win32
✔ Running preMake hook
❯ Making distributables
  ✖ Making a zip distributable for win32/x64
  ✖ Making a squirrel distributable for win32/x64
    › Failed with exit code: 1
      Output:
      Attempting to build package from 'balena_etcher.nuspec'.
      The '@' character, hexadecimal value 0x40, cannot be included in a name. Line 1, position 19.
◼ Running postMake hook

An unhandled rejection has occurred inside Forge:
Error: Failed with exit code: 1
Output:
Attempting to build package from 'balena_etcher.nuspec'.
**The '@' character, hexadecimal value 0x40, cannot be included in a name. Line 1, position 19.**
at ChildProcess.<anonymous> (/home/wu/etcher/node_modules/electron-winstaller/src/spawn-promise.ts:51:16)
    at ChildProcess.emit (node:events:517:28)
    at ChildProcess.emit (node:domain:489:12)
    at maybeClose (node:internal/child_process:1098:16)
    at Process.ChildProcess._handle.onexit (node:internal/child_process:303:5)

aethernet, 2024-05-22:
No clue, sorry 😅