balena-io/etcher #4232: ******

beatnickgr, 2024-05-26:
I copy paste the instructions under this link :

https://github.com/balena-io/etcher#debian-and-ubuntu-based-package-repository-gnulinux-x86x64

sudo apt install ./balena-etcher_******_amd64.deb
Reading package lists... Done
E: Unsupported file ./balena-etcher_******_amd64.deb given on commandline

Probably the ****** must be replaced with something else?
Please clarify the instructions.

aethernet, 2024-05-27:
it should be the file you downloaded those * represents the version

closed 2024-05-27 by aethernet

Appytoo, 2024-12-29:
rubbish unhelpful rubbish. none if it works

Appytoo, 2024-12-29:
There is no appimage! There is only a link to a download which does not work These people like to waste your time