balena-io/etcher #4239: error in the end of validation, multiple isos

kibor-mail, 2024-05-31:
- **Etcher version:** 1.19.21
- **Operating system and architecture:**  windows 10 pro 22h2 19045.4412 x64
- **Image flashed:** linuxmint-21-cinnamon-64bit.iso, linuxmint-21.3-cinnamon-64bit.iso (sha256 matches with certutil and 7-zip)
- **What do you think should have happened:** [i followed the instruction.](https://linuxmint-installation-guide.readthedocs.io/en/latest/burn.html#in-windows-mac-os-or-other-linux-distributions) i hoped it worked
- **What happened:** error in the end of validation
- **Do you see any meaningful error information in the DevTools?** 
yes. i see some errors 
[-1717123916185.log](https://github.com/user-attachments/files/15507809/-1717123916185.log)

mahendra-shah, 2024-06-05:
Same here at end of validation pop-up comes with 'something went wrong if it is a compressed image....'. 
I tried in Ubuntu and Windows with different images in both got the same problem, even tried few older versions same issue persists.

**Etcher Version:** 1.19.21
**Operating systems:** Windows 11 23H2 22631.3672, Ubuntu 22.04
**Image flashed:** Ubuntu 24.04 (.iso), Olarila Monterey (.raw)

Couldn't find any solution for this!

mkaliaha, 2024-06-09:
This is caused by a dependency that performs drive unmount (can't remember the name, will update  the comment as soon as I'll have time to reproduce again).
This depencency is compiled. In devtools there's error message that it's compiled for different version of node.js. 
The odd thing - I tried older versions that worked for me before. And it still failed. Almost like it recompiles with the node.js installed on the machine, and fails it the version is different from the one used by etcher.
I will try to downgrade my node.js installation from 20.xx to 18.xx to see if it's the case.
Per chance, do you have node.js installed?

mkaliaha, 2024-06-09:
Here's the error:

```
"The module 'C:\Users\nikol\AppData\Local\Temp\pkg\67f714fcae28480101433090de6910fead59339323ef258d9a0599cc0ea9f66c\mountutils\build\Release\MountUtils.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 93. This version of Node.js requires
NODE_MODULE_VERSION 108. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`)."
```

Library name is **mountutils**.
---
I thought it might be related to the host node.js version but downgrading from 20.xx to 18.xx or 16.xx does nothing.
So, look like the dependency wasn't rebuilt on electron's version update.
---

tangjw, 2024-10-06:
> Same here at end of validation pop-up comes with 'something went wrong if it is a compressed image....'. I tried in Ubuntu and Windows with different images in both got the same problem, even tried [...]

我升级了最新版发生了同样的问题，降级使用 balenaEtcher-Setup-1.18.11.exe，虽然该版本无法跳过校验，但是可以刷写成功！