balena-io/etcher #4240: Unable to proceed past Flash from file on Venom Linux

cngr, 2024-05-31:
With etcher 1.19.21 dosen't work distro app or appimage but with etcher 1.18.12 all work fine

cngr, 2024-05-31:
[image]

cngr, 2024-05-31:
With the old version 1.18.12 i don't have problems.

aethernet, 2024-05-31:
The internal of those two versions is completely different.
Will have a look

cngr, 2024-05-31:
appimage fails too

static-glitch, 2024-06-02:
I also get the same error with appimage on 1.18.13, 1.19.0, and 1.19.16 through 1.19.21. Appimage 1.18.12 seems to be the most recent that is working. Arch kernel 6.9.1-arch1-2

CoolManTheCool, 2024-06-02:
duplicate of [4150](https://github.com/balena-io/etcher/issues/4150) 
sollution: install V1.17.0

jstefanop, 2024-06-13:
can confirm 1.19 does not work on linux, 1.18.11 is last working version I can find