balena-io/etcher #4242: Unable to run AppImage or install on Ubuntu 24.04.

imthenachoman, 2024-06-02:
- **Etcher version:** N/A
- **Operating system and architecture:** Ubuntu 24.04 x64
- **Image flashed:** N/A
- **What do you think should have happened:** It should run/install
- **What happened:** Error
- **Do you see any meaningful error information in the DevTools?** N/A

I have tried both the AppImage and installing the deb file but I'm getting an error both ways. I am not sure what to do?

```
username@hostname:~/Downloads$ chmod +x balenaEtcher-1.19.21-x64.AppImage 

username@hostname:~/Downloads$ ./balenaEtcher-1.19.21-x64.AppImage 
[430042:0601/212033.320285:FATAL:setuid_sandbox_host.cc(158)] The SUID sandbox helper binary was found, but is not configured correctly. Rather than run without sandboxing I'm aborting now. You need to make sure that /tmp/.mount_balenaHdQcO1/usr/lib/balena-etcher/chrome-sandbox is owned by root and has mode 4755.
Trace/breakpoint trap (core dumped)

username@hostname:~/Downloads$ sudo ./balenaEtcher-1.19.21-x64.AppImage 
[sudo] password for username: 
[0601/212038.242181:FATAL:electron_main_delegate.cc(295)] Running as root without --no-sandbox is not supported. See https://crbug.com/638180.
Trace/breakpoint trap

username@hostname:~/Downloads$ sudo apt install ./balena-etcher_1.19.21_amd64.deb 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'balena-etcher' instead of './balena-etcher_1.19.21_amd64.deb'
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 balena-etcher : Depends: gconf-service but it is not installable
                 Depends: gconf2 but it is not installable
                 Depends: libgconf-2-4 but it is not installable
E: Unable to correct problems, you have held broken packages.
```

CoolManTheCool, 2024-06-02:
Stick to the app image, DO NOT USE SUDO,
`./balenaEtcher-1.19.21-x64.AppImage --no-sandbox`
IF that does't work, downgrade and you get an error when loading the ISO, switch to a stable version: [V1.17.](https://github.com/balena-io/etcher/releases/tag/v1.17.0), It works for me.

imthenachoman, 2024-06-02:
Thanks. Let me try an earlier version.

williamtorres1, 2024-06-02:
I faced the same issue, but it worked fine when i tried `sudo dpkg -i *.deb`

aliajboy, 2024-06-10:
I have same issue on Ubuntu 24.
I even tried to install it via .deb but It says I have dependency problems and can't seem to install it.

Heavygrass, 2024-07-27:
+1
Same issues on Ubuntu 24.04 all updates.

`./balenaEtcher-1.19.21-x64.AppImage 
[125230:0727/130516.296002:FATAL:setuid_sandbox_host.cc(158)] The SUID sandbox helper binary was found, but is not configured correctly. Rather than run without sandboxing I'm aborting now. You need to make sure that /tmp/.mount_balena4eK6vC/usr/lib/balena-etcher/chrome-sandbox is owned by root and has mode 4755.`

Same issues with DEB as in 1st post.

polo199, 2024-08-09:
Appimage 1.19.21 runs for me (on Kubuntu 24.04) but burning an image fails with an error message.
Appimage 1.17.0 runs fine as indicated by @CoolManTheCool .

raulgrangeiro, 2024-08-12:
> Stick to the app image, DO NOT USE SUDO, `./balenaEtcher-1.19.21-x64.AppImage --no-sandbox` IF that does't work, downgrade and you get an error when loading the ISO, switch to a stable version: [V1.17.](https://github.com/balena-io/etcher/releases/tag/v1.17.0), It works for me.
> 
> Edit: This solution _might_ work, its not recommended, but balena etcher is a VERY critical peice of tech, and there REALLY shouldn't be issues especially with (I think) the only software you can use to flash a USB.
> 
> Edit2: I keep getting off topic, and my responces aren't professional, but its enough to hopefully enough to solve your issue, also your issue is NOT because you are on 24.04, it does the same on 23.10, You probably shouldn't use anything earlier, but etcher should still have support so people can upgrade, although you dont even need a usb to install an OS, you can write the install media to your hard drive.

Your answer did work for me. The version 1.17 for some reason works fine as AppImage on Ubuntu 24.04 LTS

Thanks for this tip! God bless you!

warioishere, 2024-08-23:
I cant get balena etcher to run anymore on ubuntu 24.07
Also using version 1.17 doesnt help, on version 1.17 I get:

```
[5913:0823/210038.400539:FATAL:setuid_sandbox_host.cc(158)] The SUID sandbox helper binary was found, but is not configured correctly. Rather than run without sandboxing I'm aborting now. You need to make sure that /tmp/.mount_balenaPzaPkx/chrome-sandbox is owned by root and has mode 4755.
/tmp/.mount_balenaPzaPkx/balena-etcher: line 10:  5913 Trace/breakpoint trap   (core dumped) "${script_dir}"/balena-etcher.bin "$@"
```

on latest version it refuses to work because it says libfuse2 is not installed although fuse is definitly installed.

Any recommendations?

lazee486, 2024-09-08:
signed in to share - 
sudo sysctl -w kernel.apparmor_restrict_unprivileged_userns=0 
will temproraly allow it to run until reboot, seems to be extra security stuff on appimages...also some dependencies names have changed so the deb also fails with dpkg and apt currently.... the --no-sanbox gets it kinda running but wth glitches. remember ubuntu wants snaps...

MertGunduz, 2024-09-12:
There's a method you can use, download the file below in Balena website:

[image]

Unzip the archive, go to unzipped folder and run this command: 

```bash
./balena-etcher --no-sandbox
```

This method works on Ubuntu 24.04.

polo199, 2024-09-19:
> There's a method you can use, download the file below in Balena website...

Good solution. It works for me as well in Kubuntu 24.04. Thanks @MertGunduz

robertolabarca, 2024-10-12:
@CoolManTheCool and @MertGunduz .  _Thanks, your proposed solution worked out my problem_.
  `./netpad-0.8.0-linux-x86_64.AppImage --no-sandbox`

Xmetalfanx, 2024-10-24:
for the record i just tried the latest Appimage AND DEB on Linux Mint 22 (24.04 based) and both now install fine .. .haven't seen if they work but 1.19.25 may work now for those having issues

raulgrangeiro, 2024-10-25:
> for the record i just tried the latest Appimage AND DEB on Linux Mint 22 (24.04 based) and both now install fine .. .haven't seen if they work but 1.19.25 may work now for those having issues

I can confirm that AppImage of this newer version is working on Ubuntu 24.04 LTS if you just clock on it to execute. But, if you try to use it from terminal you'll need to use the "--no-sandbox" option. The DEB version I didn't try.

CoolManTheCool, 2024-11-08:
> you'll need to use the "--no-sandbox" option
I'm on arch linux now and the file just worked for me, using the portable version. I haven't tried the app image or anything else.