balena-io/etcher #4245: cannot install balena-etcher from aur repos using yay

vanos001, 2024-06-06:
- **Etcher version:** 1.19.17
- **Operating system and architecture:** EndeavourOS x86_64 (amd64)
- **Image flashed:** None
- **What do you think should have happened:**  it should have installed successfully
- steps  : 
- in arch based distro use : 
- ``` yay -S balena-etcher```
- **What happened:** it fails to  install balena etcher
- **Do you see any meaningful error information in the DevTools?**  None

 logs for reference :
 ```
[home@home ~]$ yay -S balena-etcher
AUR Explicit (1): balena-etcher-2:1.19.17-1
AUR Make Dependency (1): nvm-0.39.7-1
:: (1/2) Downloaded PKGBUILD: nvm
:: (2/2) Downloaded PKGBUILD: balena-etcher
  2 balena-etcher                    (Build Files Exist)
  1 nvm                              (Build Files Exist)
==> Packages to cleanBuild?
==> [N]one [A]ll [Ab]ort [I]nstalled [No]tInstalled or (1 2 3, 1-3, ^4)
==> N
  2 balena-etcher                    (Build Files Exist)
  1 nvm                              (Build Files Exist)
==> Diffs to show?
==> [N]one [A]ll [Ab]ort [I]nstalled [No]tInstalled or (1 2 3, 1-3, ^4)
==> N
==> Making package: balena-etcher 2:1.19.17-1 (Thursday 06 June 2024 08:26:22 PM)
==> Retrieving sources...
  -> Cloning etcher git repo...
Cloning into bare repository '/home/home/.cache/yay/balena-etcher/etcher'...
remote: Enumerating objects: 39890, done.
remote: Counting objects: 100% (256/256), done.
remote: Compressing objects: 100% (175/175), done.
remote: Total 39890 (delta 104), reused 166 (delta 59), pack-reused 39634
Receiving objects: 100% (39890/39890), 105.41 MiB | 3.65 MiB/s, done.
Resolving deltas: 100% (24783/24783), done.
  -> Found balena-etcher.desktop
==> WARNING: Skipping verification of source file PGP signatures.
==> Validating source files with sha256sums...
    etcher ... Passed
    balena-etcher.desktop ... Passed
==> Making package: nvm 0.39.7-1 (Thursday 06 June 2024 08:27:00 PM)
==> Retrieving sources...
  -> Downloading nvm-0.39.7.tar.gz...
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100  322k  100  322k    0     0  57926      0  0:00:05  0:00:05 --:--:-- 73743
  -> Found init-nvm.sh
  -> Found install-nvm-exec
==> WARNING: Skipping verification of source file PGP signatures.
==> Validating source files with sha256sums...
    nvm-0.39.7.tar.gz ... Passed
    init-nvm.sh ... Passed
    install-nvm-exec ... Passed
:: Remove make dependencies after install? [y/N] y
:: (1/2) Parsing SRCINFO: balena-etcher
:: (2/2) Parsing SRCINFO: nvm
==> Making package: nvm 0.39.7-1 (Thursday 06 June 2024 08:27:11 PM)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> Retrieving sources...
  -> Found nvm-0.39.7.tar.gz
  -> Found init-nvm.sh
  -> Found install-nvm-exec
==> Validating source files with sha256sums...
    nvm-0.39.7.tar.gz ... Passed
    init-nvm.sh ... Passed
    install-nvm-exec ... Passed
[... 183 lines cut ...]
 -> error making: balena-etcher-exit status 4
checking dependencies...

Package (1)  Old Version  Net Change

nvm          0.39.7-1      -0.14 MiB

Total Removed Size:  0.14 MiB

:: Do you want to remove these packages? [Y/n] 
:: Processing package changes...
(1/1) removing nvm                                 [----------------------] 100%

Don't forget to clean up any lines added to your shell's startup script!

For example, from your .bashrc (or .zshrc etc.), delete the line:

source /usr/share/nvm/init-nvm.sh

:: Running post-transaction hooks...
(1/1) Arming ConditionNeedsUpdate...
 -> Failed to install the following packages. Manual intervention is required:
balena-etcher - exit status 4
[home@home ~]$ 
```

vanos001, 2024-06-06:
found out about --debug and -v in yay 
so terminal output with debug and verbose mode : 
```
[home@home ~]$ yay --debug -v -S balena-etcher
[DEBUG:runtime] packages to query [balena-etcher]
[DEBUG:runtime] rpc request https://aur.archlinux.org/rpc?arg%5B%5D=balena-etcher&type=info&v=5
[DEBUG:grapher] deps to find [nvm]
[DEBUG:runtime] packages to query [nvm]
[DEBUG:runtime] rpc request https://aur.archlinux.org/rpc?arg%5B%5D=nvm&type=info&v=5
AUR Explicit (1): balena-etcher-2:1.19.17-1
AUR Make Dependency (1): nvm-0.39.7-1
:: PKGBUILD up to date, skipping download: balena-etcher
:: PKGBUILD up to date, skipping download: nvm
[DEBUG:runner] capturing /usr/bin/git -C /home/home/.cache/yay/balena-etcher reset --hard HEAD
[DEBUG:runner] capturing /usr/bin/git -C /home/home/.cache/yay/balena-etcher merge --no-edit --ff
[DEBUG:runner] capturing /usr/bin/git -C /home/home/.cache/yay/nvm reset --hard HEAD
[DEBUG:runner] capturing /usr/bin/git -C /home/home/.cache/yay/nvm merge --no-edit --ff
[DEBUG:db] populating db executor package caches. sync_len 873 remote_len 1
  2 nvm                              (Build Files Exist)
  1 balena-etcher                    (Build Files Exist)
==> Packages to cleanBuild?
==> [N]one [A]ll [Ab]ort [I]nstalled [No]tInstalled or (1 2 3, 1-3, ^4)
==> A
:: Deleting (1/2): /home/home/.cache/yay/nvm
[DEBUG:runner] running /usr/bin/git -C /home/home/.cache/yay/nvm reset --hard origin/HEAD
HEAD is now at 6e8d9b7 Upgrade to v0.39.7
[DEBUG:runner] running /usr/bin/git -C /home/home/.cache/yay/nvm clean -fdx
Removing nvm-0.39.7-1-any.pkg.tar.zst
Removing nvm-0.39.7.tar.gz
:: Deleting (2/2): /home/home/.cache/yay/balena-etcher
[DEBUG:runner] running /usr/bin/git -C /home/home/.cache/yay/balena-etcher reset --hard origin/HEAD
HEAD is now at e84e5af upgpkg: balena-etcher@1.19.17
[DEBUG:runner] running /usr/bin/git -C /home/home/.cache/yay/balena-etcher clean -fdx
warning: could not open directory 'pkg/': Permission denied
Removing etcher/
Removing pkg/
Skipping repository src/etcher
Removing src/.npm_cache
Removing src/balena-etcher.desktop
Removing src/.nvm
  2 balena-etcher                    (Build Files Exist)
  1 nvm                              (Build Files Exist)
==> Diffs to show?
==> [N]one [A]ll [Ab]ort [I]nstalled [No]tInstalled or (1 2 3, 1-3, ^4)
==> N
[DEBUG:runner] running /usr/bin/makepkg --verifysource --skippgpcheck -f -Cc
==> Making package: balena-etcher 2:1.19.17-1 (Thursday 06 June 2024 09:38:59 PM)
==> Retrieving sources...
  -> Cloning etcher git repo...
Cloning into bare repository '/home/home/.cache/yay/balena-etcher/etcher'...
remote: Enumerating objects: 39890, done.
remote: Counting objects: 100% (256/256), done.
remote: Compressing objects: 100% (175/175), done.
remote: Total 39890 (delta 104), reused 167 (delta 59), pack-reused 39634
Receiving objects: 100% (39890/39890), 105.34 MiB | 3.38 MiB/s, done.
Resolving deltas: 100% (24785/24785), done.
  -> Found balena-etcher.desktop
==> WARNING: Skipping verification of source file PGP signatures.
==> Validating source files with sha256sums...
    etcher ... Passed
    balena-etcher.desktop ... Passed
[DEBUG:runner] running /usr/bin/makepkg --verifysource --skippgpcheck -f -Cc
==> Making package: nvm 0.39.7-1 (Thursday 06 June 2024 09:39:39 PM)
==> Retrieving sources...
  -> Downloading nvm-0.39.7.tar.gz...
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100  322k  100  322k    0     0   213k      0  0:00:01  0:00:01 --:--:--  900k
  -> Found init-nvm.sh
[... 1269 lines cut ...]
debug: parsing hook file /usr/share/libalpm/hooks/dbus-reload.hook
debug: parsing hook file /usr/share/libalpm/hooks/texinfo-install.hook
debug: parsing hook file /usr/share/libalpm/hooks/30-update-mime-database.hook
debug: parsing hook file /usr/share/libalpm/hooks/30-systemd-daemon-reload-user.hook
debug: parsing hook file /usr/share/libalpm/hooks/update-desktop-database.hook
:: Running post-transaction hooks...
(1/1) Arming ConditionNeedsUpdate...
debug: executing "/usr/share/libalpm/scripts/systemd-hook" under chroot "/"
debug: unregistering database 'local'
debug: freeing package cache for repository 'local'
debug: unregistering database 'endeavouros'
debug: unregistering database 'core'
debug: unregistering database 'extra'
debug: unregistering database 'multilib'
debug: call to waitpid succeeded
debug: unregistering database 'local'
debug: freeing package cache for repository 'local'
debug: unregistering database 'endeavouros'
debug: unregistering database 'core'
debug: unregistering database 'extra'
debug: unregistering database 'multilib'
 -> Failed to install the following packages. Manual intervention is required:
balena-etcher - exit status 4
```
hope it helps

sukanka, 2024-06-13:
Hello, I just fixed the aur package, can you try again?

vanos001, 2024-06-14:
installed successfully and working properly 
thank you @sukanka

closed 2024-06-14 by vanos001