balena-io/etcher #4247: Error:(0,h.requestMetadata) is not a function - Ubuntu 23.10

pinaa, 2024-06-07:
This issue was closed, but it's still broken!
I would like to build etcher from source myself to fix it.

DragonflyRobotics, 2024-06-10:
I am also having the same issue on Arch OS. 
[image]

DragonflyRobotics, 2024-06-10:
Some good news! I cloned the master branch and built the project. That error seems to go away so this issue seems to be fixed on the master branch. I am flashing HomeAssistant OS and I get this error but this is likely an issue with the image itself. I thought I will share regardless:

[image]

nattadasu, 2024-06-11:
See #4150, you can pass `--no-sandbox` param on terminal to bypass this

ivanreeve, 2024-06-21:
I'm also having the same error on Windows 10
[image]