balena-io/etcher #4249: Can not installed BelenaEtcher with "sudo apt install ***"

dyghlwg7312, 2024-06-10:
- **1.19.21** 
- **Deepin20.9 Amd64** 
- **sudo apt install balena-etcher_1.19.21_amd64.deb
[sudo] password for linwengui: 
Verification successful
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package balena-etcher_1.19.21_amd64.deb
E: Couldn't find any package by glob 'balena-etcher_1.19.21_amd64.deb'
E: Couldn't find any package by regex 'balena-etcher_1.19.21_amd64.deb'
** 
-
<Then I used another a command"sudo dpkg i ***,the installation was successful">sudo dpkg -i balena-etcher_1.19.21_amd64.deb
(Reading database ... 252682 files and directories currently installed.)
Preparing to unpack balena-etcher_1.19.21_amd64.deb ...
Unpacking balena-etcher (1.19.21) over (1.19.21) ...
Setting up balena-etcher (1.19.21) ...
chmod: cannot access '/opt/balenaEtcher/chrome-sandbox': No such file or directory
Could not parse file "/usr/share/applications/screensavers/glitchpeg.desktop": Key file contains line ?several times a second.  After a while, finds a new image to corrupt. Written by Jamie Zawinski; 2018.? which is not a key-value pair, group, or comment
Processing triggers for bamfdaemon (0.5.4.1-1+eagle) ...
Rebuilding /usr/share/applications/bamf-2.index...
Processing triggers for desktop-file-utils (0.23-4) ...
Processing triggers for mime-support (3.62) ...

j75, 2024-06-14:
I have almost the same problem on Ubuntu 24.04 - the messages are 
```
 balena-etcher : Depends: gconf-service which is a virtual package and is not provided by any available package

                 Depends: gconf2 which is a virtual package and is not provided by any available package

                 Depends: libgconf-2-4 which is a virtual package and is not provided by any available package
```

baradhili, 2024-07-17:
Same problem on Ubuntu 24.04 for me  gconf2 is no longer

lurch, 2024-07-17:
See also https://github.com/balena-io/etcher/issues?q=is%3Aissue+is%3Aopen+gconf2