balena-io/etcher #4253: Source and destination checksums do not match

wwong75, 2024-06-11:
- **Etcher version:** 1.19.21
- **Operating system and architecture:** macOS Sonoma 14.5  
- **Image flashed:** apollo-2-mcu2_30424.img.xz
- **What do you think should have happened:** 
- **What happened:** 1. Verified Checksum of apollo-2-mcu2_30424.img.xz 2.  Selected file: apollo-2-mcu2_110424.img.xz 3. Selected USB Drive Sandisk Industrial 16GB SD Card 4. Flash 5.  Flashing Complete 5. Validating 6: "Something went wrong.  If it is a compressed image, please check that the archive is not corrupted.   The writer process ended unexpectedly: Cancel or Retry 
- **Do you see any meaningful error information in the DevTools?** Source and destination checksums do not match: 435bdacfb6df9410 !== f780a791f2e86f25
- 

I verified the SHA256 checksum of the downloaded image: 
ww@Win286 Downloads % shasum -a 256 apollo-2-mcu2_110424.img.xz
6cc29f8df5756fea7e94aba901ca0a0dd2a4df637000c4f38f9f1df7ea037b65  apollo-2-mcu2_110424.img.xz
ww@Win286 Downloads % 

SHA256 Checksum
6cc29f8df5756fea7e94aba901ca0a0dd2a4df637000c4f38f9f1df7ea037b65 apollo-2-mcu2_110424.img.xz

wwong75, 2024-06-11:
ndex.js:2 {size: 15931539456, isVirtual: false, enumerator: 'DiskArbitration', logicalBlockSize: 512, raw: '/dev/rdisk4', …}blockSize: 512busType: "USB"busVersion: nulldescription: "Generic MassStorageClass Media"device: "/dev/disk4"devicePath: "IODeviceTree:/arm-io@10F00000/apcie@90000000/pci-bridge1@1/pcie-xhci@0"displayName: "/dev/disk4"enumerator: "DiskArbitration"error: nullisCard: falseisReadOnly: falseisRemovable: trueisSCSI: falseisSystem: falseisUAS: nullisUSB: trueisVirtual: falselogicalBlockSize: 512mountpoints: []partitionTableType: "mbr"raw: "/dev/rdisk4"size: 15931539456[[Prototype]]: Object
index.js:2 {name: 'Error', message: 'Source and destination checksums do not match: 435bdacfb6df9410 !== f780a791f2e86f25', stack: 'Error: Source and destination checksums do not mat…jections (node:internal/process/task_queues:95:5)', code: 'EVALIDATION'}code: "EVALIDATION"message: "Source and destination checksums do not match: 435bdacfb6df9410 !== f780a791f2e86f25"name: "Error"stack: "Error: Source and destination checksums do not match: 435bdacfb6df9410 !== f780a791f2e86f25\n    at CountingHashStream.<anonymous> (/snapshot/Users/runner/work/etcher/etcher/node_modules/etcher-sdk/build/source-destination/source-destination.js:158:36)\n    at CountingHashStream.emit (node:events:518:28)\n    at CountingHashStream.<anonymous> (/snapshot/Users/runner/work/etcher/etcher/node_modules/etcher-sdk/build/source-destination/source-destination.js:78:16)\n    at process.processTicksAndRejections (node:internal/process/task_queues:95:5)"[[Prototype]]: Object
index.js:2 Starting...
index.js:2 Starting...
index.js:2 Flash results {}
index.js:2 {}
index.js:2 Starting...
index.js:2 Error: The writer process ended unexpectedly
    at i (index.js:353:5238)
    at t.createUserError (index.js:353:6464)
    at m (index.js:140:45502)
    at index.js:140:46208
    at L.<anonymous> (index.js:140:42787)
    at L.emit (node:events:518:28)
    at e.exports.H (index.js:2:2317562)
    at e.exports.emit (node:events:518:28)
    at index.js:2:2293382
    at process.processTicksAndRejections (node:internal/process/task_queues:95:5)
(anonymous) @ index.js:2
t.logException @ index.js:140
S @ index.js:140
processTicksAndRejections @ node:internal/process/task_queues:95
index.js:2 Starting...
index.js:2 Starting...
media/icon.png:1 
        
        
       GET file:///Applications/balenaEtcher.app/Contents/Resources/app.asar/.webpack/renderer/main_window/media/icon.png net::ERR_FILE_NOT_FOUND
index.js:2 Verifying 1 device, 100% at 95.50 MB/s (total 95.50 MB/s) eta in 0s with 0 failed devices
index.js:2 Amplitude Logger [Warn]: {code: 0, error: "Status 'unknown' provided for 2 events"}
(anonymous) @ index.js:2
e.warn @ index.js:2
e.handleResponse @ index.js:2
(anonymous) @ index.js:2

agranig, 2026-10-20:
I can still reproduce this on v2.1.6 with images that are over 5 GiB in size (and which therefore skip the decompression step and do the decompression somehow internally, breaking the flashing).