balena-io/etcher #4257: Cannot use balenaEtcher appimage on Ubuntu 24.04

j75, 2024-06-14:
- **Etcher version:** : **1.19.21**
- **Operating system and architecture:** : Ubuntu 24.04
- **Image flashed:** Ubuntu 24.04 ISO
- **What happened:** : After selecting the ISO file the following message was displayed:```Something went wrong while opening /usr/local/marian/ISO/ubuntu-24.04-desktop-amd64.iso.
Error: (0, h.requestMetadata) is not a function```
- **Do you see any meaningful error information in the DevTools?** ``` index.js:2 TypeError: (0 , h.requestMetadata) is not a function
    at index.js:140:1512
    at async W.openImageSelector (index.js:140:2626) ```

j75, 2024-06-14:
Trying to use the deb package does not work either, see https://github.com/balena-io/etcher/issues/4249

landoncr2012, 2024-06-14:
I'm getting the same thing but on windows.

Something went wrong while opening /usr/local/marian/ISO/ubuntu-24.04-desktop-amd64.iso. Error: (0, h.requestMetadata) is not a function

sk0va, 2024-06-15:
Same thing on Ubuntu 23.10 and Windows 11

KeithMyers, 2024-06-16:
Same issue here.  Had to revert to an older balenaEtcher-1.18.9-x64.AppImage to regain functionality.
[image]

jetspiking, 2024-06-24:
I also encountered this issue on Ubuntu 24.04 LTS. I have to admit the installation process is incredibly frustrating. I first encountered issues when downloading the x64 package on the website. This x64 app image points to an older version and downloads an ia32 version (32-bit). This results in a libfuse error, because the 32-bit binary is not present on the system. I did not verify the architecture in the file-name, I simply expected it to download the correct version.

I then decided to download directly from GitHub and could not figure out why Etcher was not launching. Running in terminal presented an error. As a workaround came across a solution where `--no-sandbox` could be appended (while running as sudo). Did not work either.

Then had to jump back a couple of versions and used the suggestion from @KeithMyers to finally get it working. Losing approximately forty minutes just to install a simple tool (which should be a straight forward installation process) is a bit much...

lurch, 2024-06-24:
Another duplicate of #4150 ?

ssj4maiko, 2024-07-08:
```
user@PC:~/Downloads $ sudo apt install ./balena-etcher_1.19.21_amd64.deb 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'balena-etcher' instead of './balena-etcher_1.19.21_amd64.deb'
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 balena-etcher : Depends: gconf-service but it is not installable
                 Depends: gconf2 but it is not installable
                 Depends: libgconf-2-4 but it is not installable
E: Unable to correct problems, you have held broken packages.
```
It seems that these packages have been deprecated in latest Ubuntu