balena-io/etcher #4258: Major Issues on all Platforms on 1.19.x releases

jstefanop, 2024-06-14:
- **Etcher version: All recent 1.19.x releases
- **Operating system and architecture:All systems
- **Image flashed: .img.xz files
- **What do you think should have happened:Proper flash and verification
- **What happened: Mac imaging fails with write process issue on .xz images, linux cant even load image into etcher

You guys need to revert your public downloads to v 1.18.11 which is the last stable version. Recent release does not work for most images on almost all platform, we are directing all our users to revert to 1.18.11

sukanka, 2024-06-16:
Hello, I tried to write [BETA-NOT-FOR-PRODUCTION-CentOS-7-aarch64-beta.img.xz](https://buildlogs.centos.org/centos/7/isos/aarch64/BETA-NOT-FOR-PRODUCTION-CentOS-7-aarch64-beta.img.xz) with self-buillt etcher 1.19.21 (using PKGBUILD from aur) on archlinux, and it works fine.

KeithMyers, 2024-06-16:
Same issue with the  .iso or . img or .xz images on Ubuntu 24.04
Can't load image just produces and error message.
[image]

Dependency issue with the .deb file.  Missing gconf and libgconf-2-4 libraries.

strophy, 2024-06-19:
I'm also experiencing this in the same way as @KeithMyers under Ubuntu 24.04 while trying to run Etcher 1.19.21 on amd64:

.deb file fails to install due to:
```
The following packages have unmet dependencies:
 balena-etcher : Depends: gconf-service but it is not installable
                 Depends: gconf2 but it is not installable
                 Depends: libgconf-2-4 but it is not installable
E: Unable to correct problems, you have held broken packages.
```
AppImage fails with:
[image]
And possibly related terminal output while trying to open the file I want to flash:
```
[479530:0619/163910.818673:ERROR:browser_main_loop.cc(280)] GLib-GObject: invalid cast from 'GtkFileChooserNative' to 'GtkWidget'
[479530:0619/163915.383284:ERROR:browser_main_loop.cc(280)] GLib-GObject: ../../../gobject/gsignal.c:2685: instance '0x1ee4019c3880' has no handler with id '12797'
```

Exorcism0666, 2024-06-21:
same issue on windows: 
[image]

digiblur, 2024-06-24:
> * **Etcher version: All recent 1.19.x releases
> * **Operating system and architecture:All systems
> * **Image flashed: .img.xz files
> * **What do you think should have happened:Proper flash and [...]

I can confirm as well.  I've seen several users trying to write the HAOS image to NVME drives and it fails.  1.18.11 works fine.

lurch, 2024-06-24:
See #4150

lukechilds, 2024-06-30:
We're also suffering from the same issue: https://github.com/getumbrel/umbrel/issues/1843

Potentially related to https://github.com/balena-io/etcher/issues/4064#issuecomment-2081366928? There has been a bug for a while where flashing compressed images from a URL does not work but from a local file does work.