balena-io/etcher #4261: Can't load any iso on Fedora 40

Noisk8, 2024-06-21:
- **Etcher version:** 1.19.21
- **Operating system and architecture:** Fedora 64bits
- **Image flashed:** everithing
- **What do you think should have happened:** 
- **What happened:** Can't load any iso
- **Do you see any meaningful error information in the DevTools?** 

[image]

[image]

D1e4Jam1e, 2024-06-21:
same for me

Noisk8, 2024-06-21:
> same for me

My solution is run a old version, exactly 1.18.11

NickJackolson, 2024-06-23:
`
Fedora 40
6.9.5-200.fc40.x86_64
`

Same here. Tried both the latest `.appimage` and `balena-etcher-1.19.21-1.x86_64.rpm`.

lurch, 2024-06-24:
Looks like a duplicate of #4150 ?

Noisk8, 2024-06-24:
🖧🖧🖧 What I did was to go to the releases section and install an old version, [version 1.8.11](https://github.com/balena-io/etcher/releases/tag/v1.18.11), download the .rpm package and install it, I was able to solve this problem in fedora 40 workstation with it