balena-io/etcher #4262: How does Etcher NOT need admin on a PC to run?

gpscruise, 2024-06-21:
I am a dick, but desperate...

Most EXE's require admin permissions to use on a work-PC.

Somehow Etcher does not.

How do you achieve that??  Not picking on you, not hacking, just advancing my knowledge.

The EXE's I write, even Windows-Forms require admin elevated privelages to run.

Thanks for any insight!

piyushyadav0191, 2024-06-26:
Etcher might be launching a separate child process with admin rights for writing and then reverting back to a standard user process.

lurch, 2024-06-26:
> Most EXE's require admin permissions to use on a work-PC.

Might be a question for your company's IT department, as to exactly what they have and haven't locked down.