balena-io/etcher #4264: Unsatisfied dependencies on kali

mahe8696, 2024-06-22:
`─(root㉿kali)-[/home/kali/Downloads]
└─#                                      
sudo apt install ./balena-etcher_1.19.21_amd64.deb  
Note, selecting 'balena-etcher' instead of './balena-etcher_1.19.21_amd64.deb'
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

Unsatisfied dependencies:
 balena-etcher : Depends: gconf-service but it is not installable
                 Depends: gconf2 but it is not installable
                 Depends: libgconf-2-4 but it is not installable
Error: Unable to correct problems, you have held broken packages.
                                                                                
┌──(root㉿kali)-[/home/kali/Downloads]
└─# apt --fix-broken install
Summary:                        
  Upgrading: 0, Installing: 0, Removing: 0, Not Upgrading: 0
`

mahe8696, 2024-06-22:
neofetch
..............                                     root@kali 
            ..,;:ccc,.                             --------- 
          ......''';lxO.                           OS: Kali GNU/Linux Rolling x 
.....''''..........,:ld;                           Host: ROG Strix G513RC_G513R 
           .';;;:::;,,.x,                          Kernel: 6.8.11-amd64 
      ..'''.            0Xxoc:,.  ...              Uptime: 17 mins 
  ....                ,ONkc;,;cokOdc',.            Packages: 3056 (dpkg), 9 (sn 
 .                   OMo           ':ddo.          Shell: zsh 5.9 
                    dMc               :OO;         Resolution: 1920x1080 
                    0M.                 .:o.       DE: GNOME 46.1 
                    ;Wd                            WM: Mutter 
                     ;XO,                          WM Theme: Kali-Dark 
                       ,d0Odlc;,..                 Theme: adw-gtk3-dark [GTK2/3 
                           ..',;:cdOOd::,.         Icons: Flat-Remix-Blue-Dark  
                                    .:d;.':;.      Terminal: gnome-terminal 
                                       'd,  .'     CPU: AMD Ryzen 7 6800H with  
                                         ;l   ..   GPU: AMD ATI Radeon 680M 
                                          .o       GPU: NVIDIA GeForce RTX 3050 
                                            c      Memory: 2480MiB / 15249MiB 
                                            .'
                                             .

lurch, 2024-06-24:
See also https://github.com/balena-io/etcher/issues?q=is%3Aissue+libgconf