balena-io/etcher #4268: (0 h.request metadata) is not a function

greatelmo, 2024-07-01:
- **Etcher version:** 1.19.21
- **Operating system and architecture:** Windows 11 x64
- **Image flashed:** linuxmint-21.3-xfce-64bit.iso
- **What do you think should have happened:** 
- **What happened:** It errored out
- **Do you see any meaningful error information in the DevTools?** No.

RupomChowdhury, 2024-07-03:
Same issue here.
Unable to flash Ubuntu Server 24.04 (Noble) and Ubuntu Desktop.
Error: (0 h.request metadata) is not a function

MurtazaKolachi, 2024-07-03:
Same issue on Win 11 pro x64 I also want Linux Mint 21.3 on my system

lurch, 2024-07-04:
See #4150

Nobody-Zhang, 2024-08-07:
> I ran it again with run as administrator and it worked for me

It works! Thanks a lot!

tliron, 2024-09-02:
Same issue, Fedora 40, Etcher 1.19.22

7Johnsz, 2024-10-07:
Guys, I’d like to share this. it worked for me, and I hope it works for you!

[image]

SirECMG, 2025-04-05:
https://forums.balena.io/t/error-0-h-requestmetadata-is-not-a-function/371138/9

It looks like it worked for balenaEtcher-1.18.11-x64.AppImage