balena-io/etcher #4278: ISO file of Linux not loading and Balena Crashing

AARUSH-VINOD, 2024-07-16:
- **Etcher version: 1.19.21** 
- **Operating system and architecture: Windows 11 x64** 
- **Image flashed: LINUX UBUNTU operating system image file** 
- **What do you think should have happened: I don't know the reason why it did that** 
- **What happened: I uploaded an ISO file of Linux ubuntu on Balena but the Flash from file option either gets stuck or if I try another way it crashes with a white background together with the whole OS and other programs on it **

12kc, 2024-07-17:
Having the same issue

maizetm, 2024-07-17:
I am also having the same problem--balenaEtcher gets suck after I select "flash from file" and choose my ISO file. I still have the option to click "flash from URL" or "clone drive", but otherwise the program is stuck.

12kc, 2024-07-17:
Also freezes when I try to flash from URL

VKCsh, 2024-07-18:
Hi folks! I was also having this issue on a Windows 11 box. I resolved the problem by installing the Visual C++ redistributable- I've confirmed this bug now on two devices.

Assuming you're up-to-date on x64 Windows 11, `winget` might solve your problem with this command:
```
winget install Microsoft.VCRedist.2015+.x64
```

You may also be able to get the redistributable from the Microsoft website, but when I test Windows I try to stick to `winget` for this sort of thing, so I'm not sure where that comes from.

I hope that helps- assuming it's the missing dependency it would be nice if BalenaEtcher could fail and warn you to install the missing components.

MandarinGris, 2024-07-23:
Also worked for me on Win10

stanimir1990, 2024-12-03:
> Hi folks! I was also having this issue on a Windows 11 box. I resolved the problem by installing the Visual C++ redistributable- I've confirmed this bug now on two devices.
>
> Assuming you're [...]

Great fix, worked for me on Win10. Thank you!