balena-io/etcher #4283: Feature Request: Embed repository for updates in DEB and RPM package

amethystgosling, 2024-07-25:
... with AppStream data so once the repository is added by installing the RPM package for example, it shows up as an app in Linux software centers and can be uninstalled from there and show up as an app in the updates tab instead of being included in the "System Updates".

lurch, 2024-07-26:
The problem is that there _isn't_ currently a "repository for updates in DEB and RPM" - see the sticky issue #4059

lurch, 2024-10-07:
> How about publishing the app as a flatpak?

:man_shrugging: https://github.com/balena-io/etcher/issues?q=is%3Aissue+flatpak+sort%3Arelevance-desc

(just to clarify: I'm not linked to this project, I'm just a helpful community member)