balena-io/etcher #4284: deb install ubuntu 24.04 ended up wrong, balena doesnt start

hanscees, 2024-07-26:
- **1.7.9** 
- **6.8.0-38-generic #38-Ubuntu SMP PREEMPT_DYNAMIC Fri Jun  7 15:25:01 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux
** 
Notice I have upgraded to 24.04 recenty from another version

- **What do you think should have happened:** 

I first clicked on icon: nothing happened
on commandline balena-etcher-electron did nothing either

- **What happened:**
nothing happened, except errors. 

so some troubleshooting:
```
balena-etcher-electron
balena-etcher-electron: command not found

$ whereis balena-etcher-electron
balena-etcher-electron: /usr/bin/balena-etcher-electron

$ /usr/bin/balena-etcher-electron
bash: /usr/bin/balena-etcher-electron: No such file or directory

ls -all /usr/bin/balena-etcher-electron
lrwxrwxrwx 1 root root 50 apr 25  2022 /usr/bin/balena-etcher-electron -> /opt/balena-etcher-electron/balena-etcher-electron

$ /opt/balena-etcher-electron/balena-etcher-electron
bash: /opt/balena-etcher-electron/balena-etcher-electron: No such file or directory

$ ls -all /opt/balena-etcher-electron/balena-etcher-electron
ls: cannot access '/opt/balena-etcher-electron/balena-etcher-electron': No such file or directory

$ ls -all /opt/
total 32
drwxr-xr-x  6 root root 4096 mei 13  2023 .
drwxr-xr-x 23 root root 4096 jun  2 17:16 ..
drwxr-xr-x  6 root root 4096 mei 13  2023 Ardour-7.4.0
-rwxr-xr-x  1 root root 4930 mei 13  2023 Ardour-7.4.0.uninstall.sh
drwxr-xr-x  5 root root 4096 apr 25  2022 balenaEtcher
drwxr-xr-x  3 root root 4096 nov 21  2021 brave.com
drwxr-xr-x  5 root root 4096 dec  9  2022 sublime_text
```

the above shows links are placed wrong somehow.

So after finding the right binary, still some sandbox thing was wrong:
```
/opt/balenaEtcher/balena-etcher-electron
[10244:0726/100342.062330:FATAL:setuid_sandbox_host.cc(158)] The SUID sandbox helper binary was found, but is not configured correctly. Rather than run without sandboxing I'm aborting now. You need to make sure that /opt/balenaEtcher/chrome-sandbox is owned by root and has mode 4755.
/opt/balenaEtcher/balena-etcher-electron: line 10: 10244 Trace/breakpoint trap   (core dumped) "${script_dir}"/balena-etcher-electron.bin "$@"

$ sudo chmod 4755 /opt/balenaEtcher/chrome-sandbox

Now it runs
/opt/balenaEtcher/balena-etcher-electron
[Axios v1.7.9] Transitional option 'clarifyTimeoutError' has been deprecated since v1.0.0 and will be removed in the near future
[Axios v1.7.9] Transitional option 'forcedJSONParsing' has been deprecated since v1.0.0 and will be removed in the near future

``` 

So I fixed the link, but thought I would let you know.

jakeuj, 2024-07-30:
> $ sudo chmod 4755 /opt/balenaEtcher/chrome-sandbox

Work for me, too.

`balenaEtcher-linux-x64-1.19.21`

```bash
sudo chmod 4755 chrome-sandbox
chmod +x balena-etcher
balena-etcher --no-sandbox
```

Thanks.

Angular-Angel, 2024-08-19:
I did all of the above, but it still doesn't work for me:

```
angle@Sager:~/Downloads$ balena-etcher --no-sandbox
d {
  _notifyingListeners: false,
  _scopeListeners: [ [Function: o] ],
  _eventProcessors: [],
  _breadcrumbs: [
    {
      timestamp: 1724094187.978,
      category: 'console',
      level: 'log',
      message: 'd {\n' +
        '  _notifyingListeners: false,\n' +
        '  _scopeListeners: [ [Function: o] ],\n' +
        '  _eventProcessors: [],\n' +
        '  _breadcrumbs: [],\n' +
        '  _attachments: [],\n' +
        '  _user: {},\n' +
        '  _tags: {},\n' +
        '  _extra: {},\n' +
        '  _contexts: {},\n' +
        '  _sdkProcessingMetadata: {},\n' +
        '  _propagationContext: {\n' +
        "    traceId: '6102b2bfa54e4142a2b4b4e95bed6ce1',\n" +
        "    spanId: '8a92daff42a46f86'\n" +
        '  },\n' +
        '  _client: k {\n' +
        '    _options: {\n' +
        '      _metadata: [Object],\n' +
        '      ipcMode: 3,\n' +
        '      getSessions: [Function: getSessions],\n' +
        "      dsn: 'https://739bbcfc0ba4481481138d3fc831136d@o95242.ingest.sentry.io/4504451487301632',\n" +
        '      beforeSend: [Function (anonymous)],\n' +
        '      debug: false,\n' +
        "      release: 'balenaEtcher@1.19.21',\n" +
        "      environment: 'production',\n" +
        '      autoSessionTracking: false,\n' +
        '      stackParser: [Function (anonymous)],\n' +
        '      defaultIntegrations: [Array],\n' +
        '      transport: [Function (anonymous)],\n' +
        "      instrumenter: 'sentry',\n" +
        '      integrations: [Array],\n' +
        '      transportOptions: [Object],\n' +
        "      platform: 'node',\n" +
        '      runtime: [Object],\n' +
        "      serverName: 'Sager'\n" +
        '    },\n' +
        '    _integrations: {\n' +
        '      SentryMinidump: [Object],\n' +
        '      ElectronBreadcrumbs: [Object],\n' +
        '      Net: [Object],\n' +
        '      MainContext: [Object],\n' +
        '      ChildProcess: [Object],\n' +
        '      OnUncaughtException: [Object],\n' +
        '      PreloadInjection: [Object],\n' +
        '      AdditionalContext: [Object],\n' +
        '      Screenshots: [Object],\n' +
        '      RendererProfiling: [Object],\n' +
        '      InboundFilters: [Object],\n' +
        '      FunctionToString: [Object],\n' +
        '      LinkedErrors: [Object],\n' +
        '      RequestData: [Object],\n' +
        '      Console: [Object],\n' +
        '      Http: [Me],\n' +
        '      Undici: [_t],\n' +
        '      OnUnhandledRejection: [Object],\n' +
        '      ContextLines: [Object],\n' +
        '      LocalVariables: [Object],\n' +
[... 124 lines cut ...]
  },
  _session: {
    sid: 'f371d636a1944210ac5ed9a49c0d0c77',
    init: true,
    timestamp: 1724094187.9746594,
    started: 1724094187.9745708,
    duration: 0.00008869171142578125,
    status: 'ok',
    errors: 0,
    ignoreDuration: false,
    toJSON: [Function: toJSON],
    did: undefined,
    release: 'balenaEtcher@1.19.21',
    environment: 'production'
  }
}

(process:44248): Gtk-WARNING **: 13:03:07.986: This process is currently running setuid or setgid.
This is not a supported use of GTK+. You must create a helper
program instead. For further details, see:

    http://www.gtk.org/setuid.html

Refusing to initialize GTK+.
```

baradhili, 2024-08-26:
The appimage works though