balena-io/etcher #4285: gconf-service dependency error when installing

skoenemann, 2024-07-26:
- **Etcher version:** 
- **Operating system and architecture:** 
- **Image flashed:** 
- **What do you think should have happened:** 
- **What happened:**
- **Do you see any meaningful error information in the DevTools?** 

Linux Mint 22.0 amd64
HP ProBook G350 

Used Gdeb file installer... have also tried command line installation. Both installation  attempts produce errors related to the gconf-service not being a resolvable dependency

[image]

[image]

lurch, 2024-07-29:
See also https://github.com/balena-io/etcher/issues?q=is%3Aissue+is%3Aopen+libgconf

skoenemann, 2024-08-13:
This dependency issue still exists in version 1.19.22, which is in "Pre-release".

skoenemann, 2024-08-13:
More research turned up this "workaround". I tried it and it installed correctly after creating a .deb file from a .rpm file using the "alien" utility...

See "https://github.com/balena-io/etcher/issues/4144#issuecomment-2156602606" and simply substitute the current version of etcher for the one in the workaround.